victronenergy/venus #1223: Check smartshunt dc energy metering modes in all UIs

mpvader, 2024-02-15:
And systemcalc.

including here:
[image]

ReinvdZee, 2024-10-25:
The Smartshunt DC meter type can be set in VictronConnect (Settings -> Misc). I've tested all DC meter types, and marked the names which are different than the names listed in VC (first column):

| Setting in VC | gui-v1 | gui-v2 | MFD app | dbus | 
|----------|----------|-----------|-------------|---------|
| Solar charger | Solar charger | Solar charger | Solar charger  | `com.victronenergy.solarcharger` |
| Wind charger | **Wind generator** | Wind charger | **Wind generator** | `com.victronenergy.dcsource` with `/Settings/MonitorMode` = -8 | 
| Shaft generator | Shaft generator | Shaft generator | - | `com.victronenergy.dcsource` with `/Settings/MonitorMode` = -7 | 
| Alternator | Alternator | Alternator | Alternator | `com.victronenergy.alternator` | 
| Fuel cell | Fuel cell | Fuel cell | - | `com.victronenergy.fuelcell` | 
| Water generator | Water generator | Water generator | - | `com.victronenergy.dcsource` with `/Settings/MonitorMode` = -4 | 
| DC/DC charger | **Dcdc charger** | **DC charger** | - | `com.victronenergy.dcsource` with `/Settings/MonitorMode` = -3 | 
| AC charger | Ac charger | Ac charger | - | `com.victronenergy.dcsource` with `/Settings/MonitorMode` = -2 | 
| Generic Source | Generic source | **DC generator** | - | `com.victronenergy.dcsource` with `/Settings/MonitorMode` = -1 | 
| Generic Load | Generic load | **DC generator** | - | `com.victronenergy.dcload` with `/Settings/MonitorMode` = 1 |
| Electric Drive | Motor | **Smartshunt** | - | `com.victronenergy.motordrive` |
| Fridge | Fridge | **DC generator** | - | `com.victronenergy.dcload` with `/Settings/MonitorMode` = 3 |
| Water Pump | Water pump | **DC generator** | - | `com.victronenergy.dcload` with `/Settings/MonitorMode` = 4 |
| Bilge Pump | Bilge pump | **DC generator** | - | `com.victronenergy.dcload` with `/Settings/MonitorMode` = 5 |
| DC System | DC system | DC system | **DC Loads** | `com.victronenergy.dcsystem` |
| Inverter | Inverter | **DC generator** | - | `com.victronenergy.dcload` with `/Settings/MonitorMode` = 7 |
| Water Heater | Water heater | **DC generator** | - | `com.victronenergy.dcload` with `/Settings/MonitorMode` = 8 |

Another observation:
Known DC meter types will remain visible in the MFD app. For example, set the DC meter type to "Solar charger", then to "Wind charger", followed by "Alternator". Now, all three types are visible in the MFD app, all pointing to the same Smartshunt.

[image]

Example comparison between gui-v1 and v2 (Smartshunt set to "Fridge"):
[image]

ReinvdZee, 2024-10-25:
The Smartshunt DC meter type can be set in VictronConnect (Settings -> Misc). I've tested all DC meter types, and marked the names which are different than the names listed in VC (first column). A dash means the name is not showing up in the UI.

| Setting in VC | gui-v1 | gui-v2 (local/WASM) | MFD app | VRM | dbus | 
|----------|----------|-----------|-------------|-------|---------|
| Solar charger | Solar charger | Solar charger | Solar charger | Solar charger | `com.victronenergy.solarcharger` |
| Wind charger | **Wind generator** | Wind charger | **Wind generator** | **DC source** | `com.victronenergy.dcsource` with `/Settings/MonitorMode` = -8 | 
| Shaft generator | Shaft generator | Shaft generator | - | **DC Source** | `com.victronenergy.dcsource` with `/Settings/MonitorMode` = -7 | 
| Alternator | Alternator | Alternator | Alternator | Alternator | `com.victronenergy.alternator` | 
| Fuel cell | Fuel cell | Fuel cell | - | Fuel Cell |  `com.victronenergy.fuelcell` | 
| Water generator | Water generator | Water generator | - | **DC Source** |`com.victronenergy.dcsource` with `/Settings/MonitorMode` = -4 | 
| DC/DC charger | **Dcdc charger** | **DC charger** | - | **DC Source** | `com.victronenergy.dcsource` with `/Settings/MonitorMode` = -3 | 
| AC charger | Ac charger | Ac charger | - | **DC Source** |`com.victronenergy.dcsource` with `/Settings/MonitorMode` = -2 | 
| Generic Source | Generic source | **DC generator** | - | **DC Source** |`com.victronenergy.dcsource` with `/Settings/MonitorMode` = -1 | 
| Generic Load | Generic load | **DC generator** | - | **DC Load** | `com.victronenergy.dcload` with `/Settings/MonitorMode` = 1 |
| Electric Drive* | Motor |  - | - | - | `com.victronenergy.motordrive` |
| Fridge | Fridge | **DC generator** | - | **DC Load** | `com.victronenergy.dcload` with `/Settings/MonitorMode` = 3 |
| Water Pump | Water pump | **DC generator** | - | **DC Load** | `com.victronenergy.dcload` with `/Settings/MonitorMode` = 4 |
| Bilge Pump | Bilge pump | **DC generator** | - | **DC Load** | `com.victronenergy.dcload` with `/Settings/MonitorMode` = 5 |
| DC System | DC system | DC system | **DC Loads** | DC System |`com.victronenergy.dcsystem` |
| Inverter | Inverter | **DC generator** | - | **DC Load** |`com.victronenergy.dcload` with `/Settings/MonitorMode` = 7 |
| Water Heater | Water heater | **DC generator** | - | **DC Load** |`com.victronenergy.dcload` with `/Settings/MonitorMode` = 8 |

*The DC meter type "Electric Drive" may not be tested properly because I did not have a motor. So the motor RPM and temperature fields (only visible in gui-v1) were empty. 

**Another observation:**
Known DC meter types will remain visible in the MFD app. For example, set the DC meter type to "Solar charger", then to "Wind charger", followed by "Alternator". Now, all three types are visible in the MFD app, all pointing to the same Smartshunt. After refreshing the page, the "Solar charger" and "Wind generator" were gone.

[image]

Example comparison between gui-v1 and v2 (Smartshunt set to "Fridge"):
[image]