victronenergy/venus #1227: Ruuvi alarms: disable them or make them happen only during the day?

mpvader, 2024-02-26:
See discussion below. Reason to tag this to v3.30 is that we're introducing this per v3.30: would be nice to do it one time right instead of first releasing in v3.30 and then improving in follow-up version.

Not sure who to assign this to. 

[image]

mpvader, 2024-02-26:
@jepefe , as in Slack, do you have a better solution / welcome to close this one.

jepefe, 2024-02-27:
The situation Stefanie suffered is just cause of a bug. The Ruuvi low battery notification is a Warning and not an Alarm but due to a bug in venus-platform the Buzzer is triggered by warnings.
This bug affects to all the warnings not only the ones from the ruuvi.

Warnings: Triggers soft notifications in the gui (alert), for example shows a red dot in the notifications icon.
Alarms: Makes the notifications page to pop up, and also triggers the Buzzer and the Relay.

Fixed in: (already merged by Jeroen)
https://github.com/victronenergy/venus-platform/commit/01b333e92b5ede7e56b68e452d60072299271cd9

closed 2024-02-27 by mpvader