victronenergy/venus #1232: Alarm relay manual toggle is broken

jepefe, 2024-03-01:
When the relay is configured as alarm relay, venus platform takes the control of it keeping its state in sync with alarms so when the user toggles it the relay status switches back immediately to the alarm state.

This issue was introduced when decoupling notifications from the gui.

Possible fixes:
- Make relay control class stop monitoring the relay state from systemcalc
- Add a Notifications/Relay/State path that the gui can use to override the alarm relay state

@jhofstee any preference?

jhofstee, 2024-03-04:
Well first of all, the "alarm relay on" toggles the wrong relay, /Relay/1/State instead of Relay/0/State.
Since venus-platform ignores changes of /Relay/1/State, above issue isn't applicable yet, it will be though when actually toggling the alarm relay.

Assuming the purpose of toggle is just to check if something is connected properly and since that can be tested nowadays in manual mode as well, or by toggling the polarity, perhaps we should simply remove this option?

kwindrem, 2024-03-04:
I always wondered why this alarm relay on/off existed. I thought maybe it was so you could disable the relay so no alarms triggered it but you can just as easily set the relay to manual, and that's not the way it works anyway.

I see no reason to have the function.

jepefe, 2024-03-04:
@kwindrem 
Because you reported the original bug, I've asked you on slack how you or other guimods users do use this toggle. 
You have already answered here.

I'll remove the alarm relay toggle from the gui.

closed 2024-03-05 by jhofstee