victronenergy/venus #1237: Heater device in dbus

dersch81, 2024-03-06:
To be aware of high load of electrical heater such as heat pump or MyPV heaters etc it would be great to have a heater device in dbus like ev charger. https://github.com/victronenergy/venus/wiki/dbus

Would it be possible to have a device with energy and temperature values as a dedicated device?

dersch81, 2024-03-06:
My first suggestion would be like this:

```
com.victronenergy.heater

/Ac/Energy/Forward         <-- 
/Ac/L1/Power               <-- L1 Power used (W)
/Ac/L2/Power               <-- L2 Power used (W) 
/Ac/L3/Power               <-- L3 Power used (W)
/Ac/Power                  <-- AC Power (W)
/Temperature/1        <-- Temp Sensor 1 degrees Celcius 
/TemperatureType    0=water; 1=generic
/Temperature/2        <-- Temp Sensor 2 degrees Celcius
/TemperatureType    0=water; 1=generic
/Temperature/3        <-- Temp Sensor 3 degrees Celcius
/TemperatureType    0=water; 1=generic
/HeatingTime              <-- Total heating time (seconds)
/Connected                 <-- 
/Current                   <-- Charging current actual (A)
/Mode                      <-- 0=Manual
                               1=Auto
/Model                     <-- Model, e.g. MyPV ACthor  or MyPV ELWA or HeatPump
/Position                  <-- 
/Status                    <-- 0=disconnected, 1=connected, 2=heating, 3=heated,
                               4=waiting, 5=waiting for start
          
```

hagre, 2025-06-21:
Please add this feature. (as new virtual device)