victronenergy/venus #1240: Since system is read-only now: include serial-starter.rules on /data

M66B, 2024-03-13:
Since the system is read-only now (good), there should be a way to add (include) rules to /etc/udev/rules.d/serial-starter.rules using the data partition, so the rules can survive a firmware update.

Similar to /data/conf/serial-starter.d

Ref: https://github.com/victronenergy/venus/wiki/howto-add-a-driver-to-Venus

kwindrem, 2024-03-13:
Those of us that create Venus OS modifications typically remount /root as read-write because that's the only way we can modify, for example the gui-v1 files.

Anything that would allow mods while leaving /root as read only would be a good thing but might be more complicated than is necessary.

M66B, 2024-03-13:
@kwindrem What about drivers for specific equipment? IMHO, they should survive firmware updates.

I don't think it is complicated to implement this, just include/add an extra rules file, and that's it.

kwindrem, 2024-03-13:
Any modification should survive a firmware update. Right now, we use code called from /data/rc.local to reinstall the mod after a firmware update. That's where we remount /root read-write and resize it so there's room for the mod.

My SetupHelper aid to Venus OS modifications stores information in /root so it IS purged on firmware update so reinstalls can occur.

Making it easier for mods to survive a firmware update IS a good thing, so I'm all for whatever can be done. I'm certainly in favor of the proposed change for serial-starter.

A generic mechanism to install mods without remounting /root as read-write is what I meant by "complicated". That would most likely require a major change to Venus OS, moving lots of it to /data which side steps the whole idea of the file system being read-only. 

Services are already handled in a way that allows them to be added with /root remaining read-only: /service is in a temp, volatile fs and /opt/victronenergy/service is copied there on boot. Services can be added to the latter location and maintain the read-only /root. Any other areas that would allow modification would need to be handled in a similar way. That's what I meant by complicated.

jhofstee, 2026-09-06:
As explained by Kevin, you can use /data/rc.local and remount the rootfs rw. You can do that only ones, and keep the rootfs read-only afterwards. I prefer that over integrating /data directly, since it provides an easy way for us to tell the firmware has been modified.

jhofstee, 2026-09-22:
Closing this, we don't want to add more things which can change boot behavior, /data/rc.local is capable already to make any change you would like.

closed 2026-09-22 by jhofstee