victronenergy/venus #1261: Gui: backslash is missing from list of selectable characters

mpvader, 2024-04-29:
Reported via sales. Apparently there are wifi networks with a \ in their password

mpvader, 2024-04-29:
I’ve asked Joona to check & test with gui-v2

jpetrell, 2024-04-29:
Neither gui-v1 and gui-v2do character validation on the Wi-Fi password, other than define maximum number of characters (63). Might be Venus platform side issue or that the special characters are not properly escaped. I'll test.

mpvader, 2024-04-29:
Hi @jpetrell: I don't know what process does something and where, but for gui-v1 there are two input methods when tpying in the password. One is using up and down button cycling through the characters, and there the \ is missing, so I can't enter such password. And when using the keyboard I can press the \ button as often as I want but it doesn't show up.

But: @jhofstee will look at gui-v1; pls from your side just make sure it works on gui-v2. Its also fine to make an issue for that and look at it only after this one is fixed up for gui-v1.

jpetrell, 2024-04-29:
Doesn't reproduce on gui-v2, only on gui-v1. Looks like Jeroen already fixed the issue. :+1:

closed 2024-04-30 by jhofstee