victronenergy/venus #1264: Prevent restarting- or otherwise fix gui-v1 lockups

mpvader, 2024-05-03:
in v3.31 this is already fixed:
> venus-platform: make sure gui-v1 doesn't get restarted during boot

This issue is about fixing it in v3.40, which needs some](venus-platform: make sure gui-v1 doesn't get restarted during boot) extra work / is a bit more complicated because of gui-v2.

mpvader, 2024-06-21:
@jhofstee this can be closed - right?

jhofstee, 2024-06-21:
well yes and no, the root cause (the lockup) is still there, but it should no longer happen during boot. A language change might trigger this. Not sure, since I can't reproduce this at the moment. So up to you what to do with this issue.

mpvader, 2024-06-21:
Close it - as a won’t fix since this bug has been in gui-v1 since for ever and we’re going to gui-v2 now.

closed 2024-06-21 by mpvader