victronenergy/venus #1265: Adaptive brightness makes the backlight pulse

jpetrell, 2024-05-06:
Customer (Lars F., Matthijs' contact) is reporting that adaptive brightness on his Cerbo GX v3.40~10 is pulsating:

https://github.com/victronenergy/gui-v2/assets/5200296/3b185514-d072-4bdf-a23d-ca62e266a3d9

Originally [filed for gui-v2](https://github.com/victronenergy/gui-v2/issues/1033), which reported two issues: broken settings switches (fixed in gui-v2) and pulsating display brightness (issue here). Reported pulsation here as gui-v2 doesn't implement the adaptive brightness, just provides the user setting.

mpvader, 2024-05-06:
What does @ReinvdZee need to do @jpetrell ? 

Run latest gui-v2 and see if he can reproduce it? And if not, then lets close the issue. I'll report back to Lars that he can try with a later Venus OS version.

jpetrell, 2024-05-06:
> What does @ReinvdZee need to do @jpetrell ?
> Run latest gui-v2 and see if he can reproduce it?

Yeah try to repro. Though I couldn't reproduce, and neither likely can @ReinvdZee. Also, could review the adaptive brightness code for flaws, or try to reduce likelihood of the issue by making the brightness steps smaller, add some time filtering, implement hysteresis to avoid constant jumps, etc.

> And if not, then lets close the issue

Yeah, I guess only do more investigations if more people start reporting the issue.

mpvader, 2024-05-06:
ok closed then.

closed 2024-05-06 by mpvader