victronenergy/venus #1267: dbus-modbus-client: rename /AutoStart path

philipptrenz, 2024-05-13:
Rename `com.victronenergy.genset` and  `com.victronenergy.dcgenset` path `/AutoStart` to `/RemoteStartModeEnabled` and also change wording in gui's. The reason is to have less ambiguity compared to our own autostart/stop feature.

AFAIK, it requires changes in:
- [ ] https://github.com/victronenergy/dbus-modbus-client
- [ ] https://github.com/victronenergy/dbus_generator
- [ ] https://github.com/victronenergy/gui
- [ ] https://github.com/victronenergy/gui-v2

@martinbosma asked for syncing this with https://github.com/victronenergy/venus-private/issues/316

philipptrenz, 2024-05-13:
@jepefe: Does the `dbus_generator` "Autostart mode not enabled" alert watch for the remote generator or the service to be set to "Auto start/stop"?

ReinvdZee, 2024-05-22:
So we don't forget; we should update these as well:

https://github.com/victronenergy/venus/wiki/dbus#generator-data
https://www.victronenergy.com/live/ccgx:fischer_panda#troubleshooting

mpvader, 2024-05-22:
good one. I've added it to the tracker in the google sheet

ReinvdZee, 2024-05-22:
>@jepefe: Does the dbus_generator "Autostart mode not enabled" alert watch for the remote generator or the service to be set to "Auto start/stop"?

@philipptrenz, I believe you're referring to this toast:
[image]

The toast pops up if "Mode" is set to "On", but the "Autostart" functionality on the genset is not enabled (`/AutoStart` == 0), so the GX won't be able to turn on the genset with the relay.

mpvader, 2024-05-22:
> @jepefe: Does the dbus_generator "Autostart mode not enabled" alert, see below screenshot, watch for the remote generator or the service to be set to "Auto start/stop"?

the answer to that is no. I'll make a new issue now to get that added.

ReinvdZee, 2024-05-22:
@philipptrenz, @mpvader,

The genset controller menu has the setting "Mode" ([qml](https://github.com/victronenergy/gui/blob/master/qml/PageAcInModelGenset.qml#L35)), which can currently be set to "on", "off" or "Auto start/stop":

[image]

We're not sure if the "Auto start/stop" option of the "mode" menu should be renamed as well. 

When set to "Auto start/stop", control is handed over to the `generator start/stop` by: 
1. `com.victronenergy.generator.startstop1/ManualStart` = 0 --> Manually turning off the generator relay first
and
2. `com.victronenergy.generator.startstop1/AutoStartEnabled` = 1 --> Enabling generator auto start/stop

So the genset will be controller by the auto start/stop feature of `dbus_generator`. 

I suppose this is the "Auto start/stop" feature @philipptrenz was talking about here: 
>"Auto Start/Stop" → our own start/stop software feature

If so, then this Mode item does not need renaming.

philipptrenz, 2024-05-26:
@ReinvdZee Yes, the above is our own "Auto Start/Stop" feature, so please keep the gui string as is.

ReinvdZee, 2024-05-27:
Thanks @philipptrenz for your answer. We will leave that gui string as is.

ReinvdZee, 2024-05-27:
Because there was quite some confusion about the generator start stop feature, we summarized the implementation of the auto start/stop feature in a [document](https://victronenergy-my.sharepoint.com/:w:/r/personal/rvanderzee_victronenergy_com/Documents/DBUS%20generator%20start%20%20stop.docx?d=wf44e87df27514ef39b0131e37a22a4fc&csf=1&web=1&e=ptHauU).

In addition to renaming the `/AutoStart` path, there are two other changes:

- Make the "Auto start/stop" submenu in `PageAcInModelGenset.qml` always visible.
- Change the toast text issued when mode is set to "on" but the genset cannot be remotely started to:
  "Unable to start the genset because its remote start functionality is disabled. Enable it on the genset control panel"

ReinvdZee, 2024-08-20:
Merged per Venus OS 3.50~1. Tested and verified. Closing.