victronenergy/venus #1273: DSE: add path to remotely enable autostart mode on the DSE controller

mpvader, 2024-05-22:
Work has been done by Philipp:

https://github.com/victronenergy/dbus-modbus-client/tree/philipptrenz/dse-autostart-mode-switch

But needs some changes, and please link to this issue in the commit.

- this should have a different name, now that we've renamed /Autostart. How about /RemoteStartEnable ?
- And the other thing what is missing is either comments in the code or here is something that that explains the background of that option in more detail. Ie. is there a user setting on the DSE that enables SCF_SELECT_AUTO_MODE? Or does it depends on the model.

mpvader, 2024-05-22:
this is to be assigned to @philipptrenz, but for some reason I can't

philipptrenz, 2024-05-26:
Also can't assign it to myself. 

DSE has binary flags indicating if certain functionality is available, which indeed depends on the model. The path gets only added, if the controller indicates that this functionality is available.

philipptrenz, 2024-05-26:
Decided to go with `/SetRemoteStartMode` to differentiate between new `/RemoteStartModeEnabled`, I will now update the dbus wiki entry.

mpvader, 2024-05-27:
can you both disable and enable the remote start mode via that path @philipptrenz ? And do we even want to allow disabling that from a GX device?

If not, how about slightly different: `/EnableRemoteStartMode`.

only disadvantage is that on dbus-spy they'll be far apart due to alphabetic ordering

philipptrenz, 2024-05-27:
Typically, there are more than two modes and they differ in meaning/naming between manufacturer. So it makes sense to only allow engaging of remote start mode and resetting has to be done on the genset itself, I'd say. 

I agree on `
/EnableRemoteStartMode
`, will adjust accordingly.

@mansr That's also relevant for the `mans/cre` branch on dbus-modbus-client.

philipptrenz, 2024-09-06:
`/EnableRemoteStartMode` was added to the DSE integration as ofh dbus-modbus-client v1.53, closing

closed 2024-09-06 by philipptrenz