victronenergy/venus #1274: Generator start/stop alarms: add watching the Remote start mode of genset

mpvader, 2024-05-22:
To have automatically starting working correctly, there are two conditions:

- Auto start/stop in the GX needs to be enabled.
- Remote start setting in the genset needs to be enabled.

We have an alarm feature, that monitors if the automatically starting and stopping of the generator works correctly. See below screenshot for the setting meant here.

Right now that only monitors the first, not the second. And monitoring the Remote start setting is quite wanted, so in the field its often disabled for maintenance for example, and then forgotten to enable it again. So this issue is about making it monitor also the second one.

We'll make a seperate alarm for this, so there will be two alarms:
1. Alarm: Auto start/stop on the GX is disabled.
2. Alarm: Remote start mode in the genset is disabled

Note that we're *not* adding a second user setting for this. There is one user setting, and the GX then monitors both paths.

Details:
- dbus-generator: add code to monitor, when relevant, the `/RemoteStartModeEnabled` path of the genset.
- venus-platform: add the new alarm.

[image]

ReinvdZee, 2024-05-23:
I'm trying to get the expected functionality straight, do I understand this correctly (@mpvader)?

- There is 1 user setting in "Generator start/stop": "Alarm when generator is not in auto start mode". 
- When the setting is disabled, no alarms must be triggered.
- When the setting is enabled, trigger:
    - an alarm if the GX Auto start/stop feature is disabled for 10 minutes 
    - an alarm when the remote start mode is disabled for 10 minutes

I'm asking because the alarm evaluation function was [modified](https://github.com/victronenergy/dbus_generator/commit/cc7d1bb4ce811c339ee70bf02117d341c85a0510) (@izak) to look at `/RemoteStartModeEnabled`, but its behavior is not what I'd expect.

Currently, an alarm will only be triggered when GX Auto start/stop is **enabled** AND the genset remote start mode is disabled.

izak, 2024-05-23:
@ReinvdZee We only changed the name of the path, not any of the functionality, so behaviour should have remained the same. The function is supposed to remind you to turn autostart back on, in case you turned it off because you were working on the genset. For safety, a service man will always turn autostart off before working on the machine.

It should warn you, after a timeout, at least if you forgot to re-enable autostart on the panel (because that requires sending someone out back to the site again). That is what I recall at least. @jepefe originally designed this and will know better.

mpvader, 2024-05-23:
@ReinvdZee , when that, already existing, setting is enabled, then:

- an alarm should be raised if auto start/stop on the gx is  disabled.

And also:

- an alarm should be raised if /RemoteStartModeEnabled path exists, isn’t invalid, and is 0.

And both of these alarms have a different text, to clearly tell the user what the problem is.

And now that I think of it, to have this work properly on VRM, both of them need to have their own path on dbus as well btw. With two states: 0 = OK and 2 = Alarm 

and then we also need data attributes.

jepefe, 2024-05-23:
> @ReinvdZee We only changed the name of the path, not any of the functionality, so behaviour should have remained the same. The function is supposed to remind you to turn autostart back on, in case [...]

I believe you are considering this from the perspective of generators with a panel. The functionality was originally designed for the GX generator start/stop autostart feature, not the remote panel's RemoteStartModeEnabled. However, after the change Rein referred to, this alarm mechanism has been modified to monitor the panel RemoteStartModeEnabled but not the GX autostart.

mpvader, 2024-05-23:
So we have a bug in there since Q4 2023?

jepefe, 2024-05-23:
Unfortunately, that appears to be the case.

mpvader, 2024-05-23:
ok. That does explain some confusion that I've heard now and then.

What I'd like to be done is, aside from fixing this in venus master aka, v3.40 betas, also release a Venus OS v3.33 that fixes this bug. And then preferably include checking remote start mode as well: ie check both paths as how its discussed above.

Can you take care of that @ReinvdZee , and have Jesus review it?

Please before you start on this, consider a bit how you'll do it with regards to releases; if needed to keep it practical, its even ok for me to just force push to master of dbus_generator to remove all recent changes, ie. v1.6.12. Then do these fixes; test and release that carefully.

Thereafter re-add dcgenset service and other things again.

For now, I've removed the pending entry on the todo, to reduce work on Venus OS maintainer side.

ReinvdZee, 2024-05-24:
@mpvader,

Venus v3.32 has dbus-generator v1.6.10. 

Should I then branch off of commit [26ea1c..](https://github.com/victronenergy/dbus_generator/commit/26ea1cca57fe33ed61b118663c0a8732f330f312) (the 1.6.10 tag) and do the changes there?

izak, 2024-05-24:
> Should I then branch off of commit [26ea1c..](https://github.com/victronenergy/dbus_generator/commit/26ea1cca57fe33ed61b118663c0a8732f330f312) (the 1.6.10 tag) and do the changes there?

Yes. Make a branch that branches off at the tag that was in 3.30. I usually call it `venus-3.3x` according to the current stable cycle. 

Then the idea is to first fix the issue on master, and then cherry-pick it across to this branch so you can easily make a patch.

Then when venus 3.33 is built, this is added as a patch to the build process and the release number (of the OE package) is incremented. Jeroen handles that.

A nice feature of github, once you have the changeset on the screen, add ".patch" to the end of the url. List this on the TODO, with the (**backport candidate**) tag. That makes it easy to pull a patch that can go right into the build tree.

ReinvdZee, 2024-05-24:
>@ReinvdZee , when that, already existing, setting is enabled, then:

About the setting "Alarm when generator is not in auto start mode", there is actually two (or more) ways to navigate to this setting, and both result in a different dbus path being set. 

**1: settings menu**
Device settings --> "Generator start/stop" --> "settings":

[Screencast from 24-05-24 12:13:41.webm](https://github.com/victronenergy/venus/assets/38864570/7e48a52c-96d9-4364-ae53-657cc3b4853b)

This setting sets `/Settings/Generator0/Alarms/AutoStartDisabled`

**2: Through genset menu**
genset --> "Auto start/stop" --> "Settings":

[Screencast from 24-05-24 12:17:02.webm](https://github.com/victronenergy/venus/assets/38864570/d564c2e5-11bb-4779-85a3-e5114e5ece1e)

This setting sets `/Settings/Generator1/Alarms/AutoStartDisabled`

So will the setting accessed through the genset menu be removed, or should we point it to the same dbus path? Because now, the alarm on the `/RemoteStartModeEnabled` will only trigger when the setting through the genset menu is enabled, because that instance of `startstop` is looking at the genset.

@mpvader, @jepefe

jepefe, 2024-05-24:
@ReinvdZee 

Via the two menus you are actually accessing to two independent instances of the generator start/stop:

`/Settings/Generator0/Alarms/AutoStartDisabled ` applies to the Relay generator and `/Settings/Generator1/Alarms/AutoStartDisabled` applies to the Fischer Panda generator.

Both settings must be kept.

ReinvdZee, 2024-05-27:
My plan for fixing the alarms:

Make the two separate alarms work and rename `/AutoStart` on a feature branch. Then merge onto master and cherry-pick only the separate alarms fix to a backporting branch (venus-v3.3x, branched off at tag v1.6.10). 

The backport will then have the two alarms (and therefore fix the bug), but won't have the renaming part.

@mpvader

ReinvdZee, 2024-05-27:
My plan for fixing the alarms:

Make the two separate alarms work and rename `/AutoStart` on a feature branch. Then merge onto master and cherry-pick only the separate alarms fix to a backporting branch (venus-v3.3x, branched off at tag v1.6.10). 

The backport will then have the two alarms (and therefore fix the bug), but won't have the renaming part. 

To have the new alarm actually show up, a change to venus-platform is needed as well. Same here, do changes (add 1 line [here](https://github.com/victronenergy/venus-platform/blob/master/src/alarm_item.cpp#L49)) on master and cherry-pick over to a backporting branch. 

So the backport will have the two alarms and is two patches, ok?
@mpvader

mpvader, 2024-05-27:
hi Rein, yes, the fix can be as many patches as you want. 👍

mpvader, 2024-05-28:
Here are the two alarms, really nice how they work now:

[image]

mpvader, 2024-05-28:
I think there is still some work on vrmlogger, and actually another bug: the old alarm was never made available to VRM (!!)

mpvader, 2024-05-28:
@ReinvdZee / @izak see updated todo list at the top. And Izak pls confirm that my suspicion is correct, that we have never added this to VRM.

ReinvdZee, 2024-05-30:
Alarms on gui-v1:

[image]

And on gui-v2:
[image]

closed 2024-06-08 by mpvader