victronenergy/venus #1278: Add support to NMEA2000 Tanks sender with different class/function

macjl, 2024-05-28:
Hello,

As discussed [here](https://community.victronenergy.com/questions/288213/kus-nkc42-nmea2000-not-shown.html), there are N2k tanks senders that do not have a class/functions of 80/190 or 80/170 or 75/150, as [required](https://www.victronenergy.com/media/pg/Cerbo_GX/en/connecting-supported-non-victron-products.html#UUID-9f759229-3e91-5a02-891c-2206c9625753).

Would it be possible to add the class/function 25/132, to allow KUS NKC42 to send tanks data to VenusOS ?

mpvader, 2024-07-23:
Hey,

This is in per Venus OS v3.50~4. (a beta version).

To test it, keep an eye out on the announcements here: https://community.victronenergy.com/spaces/31/mods.html

And once you have that installed, can you confirm that it works @macjl ?

@martinbosma : have we tested this?

martinbosma, 2024-07-23:
@mpvader I did not test this on a real device, but I have made a simple n2k tank simulator and confirmed that it worked with the class/function of the KUS NKC42. If you think it is needed, I can purchase the real device.

macjl, 2024-07-23:
I’ll test it as soon as it will be available, and let you know if it works.

mpvader, 2024-07-23:
> @mpvader I did not test this on a real device, but I have made a simple n2k tank simulator and confirmed that it worked with the class/function of the KUS NKC42. If you think it is needed, I can purchase the real device.

I don’t think thats needed. Lets first await test by @macjl

macjl, 2024-07-25:
Hello,

I can confirm it works now.

Thank you !

mpvader, 2024-07-25:
great! thank you.

(ping fyi @martinbosma )

closed 2024-07-25 by mpvader