victronenergy/venus #1283: Parallel Lynx BMS: Show that Lynx BMS is waiting to connect to a parallel system

ReinvdZee, 2024-05-31:
Since Lynx BMS firmware v1.10-beta03, it is possible to parallel multiple Lynx Smart BMSes. A Lynx Smart BMS can only connect to the system safely when the voltage difference between system and battery bank (voltage across contactor) is sufficiently small. When the BMS powers up, it checks the voltage difference and remains in a waiting state until the voltage difference is small enough. 

currently, the gui displays `[soc voltage current]` as summary on the device page:
[image]

When in the waiting state, the current will just be 0A, so it is not very clear that the BMS is actually waiting to connect. 
So we should change the summary on the device page to show something like "Waiting" or "Standby".

ReinvdZee, 2024-05-31:
There are two dbus paths reflecting the state of the BMS: 

- `/State`: Generic state like "Running", "Shutdown", "Initializing" etc.
- `/DeviceState`: The `VE_REG_STATE`, reflects the charging state, amongst others. (used to check if the BMS is in Float or not for the parallel BMS)

When the BMS is in the waiting state, `/State` is 16 and `/DeviceState` is 249. `/State` is also displayed in the battery page:
[image]

Which thus displays "Pre-charging" when the BMS waits until it can connect. 

Shouldn't the device-page then also display "Pre-charging"? Or would displaying "Standby" on the device list and displaying "Pre-charging" on the device-page be fine?

@mpvader

mpvader, 2024-06-14:
what did we decide for this @ReinvdZee ?

This is an urgent one, I want to release v3.40 asap, and support for parallel lynx bms is part of that.

ReinvdZee, 2024-06-17:
@mpvader We didn't decide on anything yet, we also need Roland's opinion on this. I'll start a thread in slack

ReinvdZee, 2024-06-17:
@mpvader, 

What to display on the GUI for the BMS that is waiting to connect to the system?  
- "Standby" suggests the user has turned off something and needs to turn it back on.  
- "Syncing" suggests the BMS is actively doing something before it can connect.  
- "Pre-charging" is too technical.  

We decided:  

- Gui-v1: "Waiting" + system voltage  
- Gui-v2: "Waiting" + voltage delta + threshold

I'll make the changes for V1 and create an issue for V2.

ReinvdZee, 2024-06-20:
Gui-v1 result:

[image]

mpvader, 2024-06-21:
@ReinvdZee yes.

ReinvdZee, 2024-06-21:
Updated the devicepage:

[image]

And this is also updated:
>When the BMS is in the waiting state, `/State` is 16 and `/DeviceState` is 249. 

When the BMS is in the waiting state, `/State` will be 18 (`WAIT_FOR_PARALLEL_CONNECTION`). `/DeviceState` is unchanged (`VE_REG_STATE_ON`).

ReinvdZee, 2024-06-21:
For the gui-v2 part, we need the voltage threshold set by the parallel BMS. Currently, the GX does not know this value. 

A [VE_REG](https://github.com/victronenergy/velib/commit/a1c1420a6b057b98984b311e3ef61d72ac3733b0) has been added for this. dbus-vecan [needs to support it](https://github.com/victronenergy/venus-private/issues/392) before gui-v2 can use it.

ReinvdZee, 2024-06-21:
I moved the V2 work to a separate issue: https://github.com/victronenergy/gui-v2/issues/1269

mpvader, 2024-06-21:
Hey @ReinvdZee , wrt 

> Gui-v2: "Waiting" + voltage delta + threshold

I'm probably changing my mind: I'd like to make it just show the battery bank voltage (which btw is not the same as the system voltage!), instead of the earlier plan, delta + threshold.

The reason is that battery voltage is simpler.

ReinvdZee, 2024-06-24:
@mpvader 

So then we just show the battery bank voltage on the summary in the devicelist on both gui V1 and gui V2? And in the devicepage:
- set state to "Waiting"
- show system voltage, only display this field when in "Waiting" state.

ReinvdZee, 2024-06-25:
@mpvader 

Like this:

[image]
Showing the waiting state and its own battery voltage. 

[image]
Also shows that its waiting. The summary entries are unchanged (own battery voltage, current & SOC). 
Field "System voltage" displays the virtual BMS voltage and is only visible when the BMS is in the waiting state.

I think this is actually more user friendly than what we've designed before. Is this what we want?

mpvader, 2024-06-25:
Looks good to me! @rjacobsVE ?