victronenergy/venus #1284: Add code to delete commonly used eternal growing logfiles

mpvader, 2024-05-31:
Recently, I've had various systems that were stuck because the data partition is full.

I'm in touch with all those people to get them to change their ways, see here for the most detailed one: https://github.com/henne49/dbus-opendtu/issues/173.

And we have an issue to actually tell the user that his disk is full.

But aside from that, and probably much less work, perhaps we should just at boot, before those services start, delete certain files if they exist.

This issue is to keep a list while I'm getting in touch with those projects.

- `/data/dbus-opendtu/current.log`, they solved it for new releases, including deleting the file when user re-runs their install script. [issue](https://github.com/henne49/dbus-opendtu/issues/173)
- `/data/dbus-shelly-em-smartmeter/current.log`, root cause discussion started today [commnity](https://community.victronenergy.com/questions/127339/shelly-1pm-as-pv-inverter-in-venusos.html)
- `/data/dbus-shelly-3em-smartmeter/current.log`, root cause not fixed yet [issue](https://github.com/fabian-lauer/dbus-shelly-3em-smartmeter/issues/69)
- '???', discussion: https://github.com/vikt0rm/dbus-shelly-1pm-pvinverter/issues/4

mpvader, 2024-05-31:
@izak fyi

mpvader, 2024-06-13:
@mansr can you add a script somewhere early at boot that removes the above listed four files? thanks.

philipptrenz, 2024-07-21:
It spreads more and more:
* https://github.com/vikt0rm/dbus-goecharger
* https://github.com/0x7878/dbus-goecharger
* https://github.com/parlue/dbus-goecharger
* https://github.com/gonzo7734/dbus-goecharger

I think we should consider `/data/dbus-*/current.log`. A simple `rm` might not work due to the file size, I like to use `cat /dev/null > current.log` for such cases.

mpvader, 2024-09-16:
discussed with Jeroen today, he'll look into it (and then most likely take the initscripts commit into master)

closed 2024-09-16 by jhofstee