victronenergy/venus #1286: Generator start/stop: Start a connected genset with GX relay

ReinvdZee, 2024-06-05:
See [this](https://victrondevelopment.slack.com/archives/C7C1ZU6UD/p1714999675875049) discussion on slack. 

Connected gensets start instace `com.victronenergy.startstop1`, which is able to start the connected genset via modbus. However, some genset models support readout via modbus, but cannot be started remotely over modbus. They can only be started with the start relay. 

To support this, there must be a way to let `startstop1` control the GX generator relay. Preferably without starting `startstop0`. 

@philipptrenz

ReinvdZee, 2024-08-05:
[Initial implementation](https://github.com/victronenergy/dbus_generator/commit/5f8c0b8e83d21e25634b83f1b9bbefa1025d71f3)

The relay must be configured as "Connected genset helper relay" in the gui:
[image]

If there is a genset detected, it will show up in the device list, as always and the helper relay state will reflect the genset state. In case the connected genset does not support starting/stopping over Modbus, the `/Start` path will be invalid and the relay must be used to control the genset. If this is the case, but the relay is not configured properly, the GUI shows the following:

[image]

And the `Autostart functionality` and `Manual start` menu items are disabled. After configuring the relay as helper relay, the genset can be used.

When - in case of a connected genset with `/Start` path - the relay is configured as helper relay while the genset is already discovered, the relay state will reflect the genset state. So configuring the relay as helper relay while the genset is running will directly engage the relay.

philipptrenz, 2024-08-20:
@ReinvdZee The DSE4520 MKII integration is ready to be included: https://github.com/victronenergy/dbus-modbus-client/tree/philipptrenz/DSE4520MKII

philipptrenz, 2024-08-20:
@ReinvdZee I reviewed the related changes and I have one note regarding https://github.com/victronenergy/dbus_generator/compare/rein/genset_runninghours...victronenergy:dbus_generator:rein/genset_relay#diff-1aaec4362ebec1eee73c529825351600806501b9f7fd0ad66865d13b2b05259cR105-R107 

Shouldn't this part only be executed if `com.victronenergy.settings/Settings/Relay/Function` equals 5?

ReinvdZee, 2024-08-21:
Changes are ready and can be put on the todo:

- dbus_generator v1.6.19
- gui, branch [rein/genset_relay](https://github.com/victronenergy/gui/tree/rein/genset_relay)
- dbus-modbus-client, branch [philipptrenz/DSE4520MKII](https://github.com/victronenergy/dbus-modbus-client/tree/philipptrenz/DSE4520MKII)

gui-v2 changes are not ready yet ([issue](https://github.com/victronenergy/gui-v2/issues/1416)), but can be added later.

@philipptrenz Will beta test this.

But before I place it on the todo, @mpvader, @philipptrenz do you agree with the UI like this? Or should we add some feedback for the user saying that the relay will be engaged as well? We could add a field "Helper relay" which only shows when the relay is configured as helper relay and set it to "enabled" or "disabled" when the generator is enabled/disabled.

philipptrenz, 2024-08-21:
@ReinvdZee I like the idea of the "Helper relay" entry in the menu, that makes it more accessible to the users. I would only opt for "open" and "closed" as state labels.

closed 2024-09-24 by jhofstee