victronenergy/venus #1287: Connected gensets: complete the UI/UX cleanup

mpvader, 2024-06-05:
In v3.33 we are making, a few changes to improve the UI. Some things were left out to not delay releasing v3.33 any further.

This issue is to list the things we need to look into as a follow-up:

- [ ] There are now two runtime counters in the menu, (and on VRM etc.). The one by the genset panel is the one we want - the counter by dbus-generator needs to be hidden or otherwise not used.
- [ ] We need to make sure that runtime + service interval as visible on VRM uses the DSE one.
- [ ] There is no runtime reset button for the DSE controller (or is there @philliptrenz ? )
- [ ] There are two error states, one in the top level menu, and one in the lower one.
- [ ] The status of the Remote start option (in the genset panel) is now not visible in the GX menus; its only visible when its wrong. Consider fixing that.
- [ ] Make sure all UIs are aligned (gui-v1, gui-v2, VRM, html5-app)

philipptrenz, 2024-06-05:
> There is no runtime reset button for the DSE controller (or is there?)
No, I think the generator runtime is like the odometer on a car, it should not be manipulable. 

One note to service interval: Some controllers have their own service interval counter with reset option, but I think it's fine that both systems provide that and it is up to the customer to decide which one to use.

mpvader, 2024-06-05:
@philipptrenz , sorry, I wasn’t clear: I meant button to reset the service interval counter. Totally agree on the total runtime counter.

Does our system already use the DSE runtime counter for its own counting? Or if not the DSE status (running/stopped) in order to when when the counter needs to count and when not?

philipptrenz, 2024-06-07:
@mpvader As far as I know, it does not use the reported runtime of the genset, but I think it initially uses the readings when a genset is first found, which still causes the values to drift apart.

philipptrenz, 2024-06-07:
I want to propose to streamline this issue with additional tasks:

- [ ] Allow to enable Remote start mode via gui → see https://github.com/victronenergy/venus/issues/1273 
- [ ] Replace "Generator stop time" for connected gensets, as we know when they are actually stopped
- [ ] Replace "Generator stop time" when state is reported by digital input (see [here](https://www.victronenergy.com/media/pg/Cerbo_GX/en/digital-inputs.html#UUID-26968630-0052-21ca-cc8f-c15551e93406))

[image]

mpvader, 2024-06-08:
One more here. And for me its not necessary to fix that in gui-v1; lets just make sure its 100% in gui-v2.

[image]

mpvader, 2024-06-09:
Hey @philipptrenz ,

> I also have a stack of VRM issues related to generator, which El Niño asked to discuss next Wednesday with me and @ReinvdZee:

I went through those issues; finally - sorry for the delay. See my responses there - hope that helps!

ReinvdZee, 2024-06-10:
I've added a todo item for the clear error gui button currently implemented. The current implementation was originally added for FP but is outdated. It does not do anything on the genset (possibly only on very old genset firmware versions, according to @martinbosma). Newer gensets support clearing errors via Modbus. 

We should offer the option to clear errors remotely only when that is actually supported by the genset.

ReinvdZee, 2024-07-16:
I moved some of the tasks to separate issues to track them separately.

philipptrenz, 2024-09-06:
Just noticed, that the error message says "Remote switch control disabled", but should now say "Remote start mode disabled". I suspect that this may have slipped through during path renaming.

[image]
[image]

ReinvdZee, 2024-10-03:
Most of the bullet points are checked. As per the UI alignment, gui-v1 and v2 won't be in sync since we're not doing all changes to v1 anymore. And the other UI-related items are tracked in different issues. (see this [one](https://github.com/victronenergy/venus/issues/1350) about VRM).

Closing this issue.

closed 2024-10-03 by ReinvdZee