victronenergy/venus #1288: Connected gensets: v3.33 changes

mpvader, 2024-06-05:
- [x] Remove the "Mode" option on the genset toplevel; which had On/Off/Auto
- [x] Move the "Autostart functionality" and "Manual start" functionality from the "Autostart" submenu up to the toplevel.
- [ ] Add menu entry that shows autostart stop system status; show only when auto start is disabled.
- [ ] Add menu entry that shows status Remote start mode. Title `Remote start mode`, which is read only.
- [ ] Reword (and keep!) the toast that warns the user if he tries to start the generator while remote start is disabled. Text: `Unable to start the genset, its Remote Start functionality is disabled. Enable it on the genset control panel.`

mpvader, 2024-06-06:
Here a picture showing the result:

[image]

mpvader, 2024-06-06:
this work continues here: https://github.com/victronenergy/venus/issues/1287

closed 2024-06-06 by mpvader

philipptrenz, 2024-06-07:
Added missing control statuses *Warm-up* and *Cool-down*:
[image]