victronenergy/venus #1290: Multi shows disconnected when switched off

kwindrem, 2024-06-15:
If the Multi (or Quattro I assume) is switched off when the GX device boots, the dbus service for the multi  /Connected parameter is 0. If you switch on the Multi then switch it off, /Connected is 1.

This is affecting the validity of the system parameter /VebusService and anything that uses it.

If the Multi is present, I would expect /Connected to be 1 even if the GX device boots with the Multi switched off.

I am tested on v3.40~26 and v3.33 but have no reason to suspect this is a new bug.

jhofstee, 2026-09-14:
Well, I don't even know if this is still an issue, but if a Multi is switched off, it is really off, as in powerless. The original code would leave it off, if it is known that it is turned off. It is not that surprising (for me) that connected is 0, since there is no communication on purpose. When you tell it to go on, it will be powered and there will be communication, likewise to tell it to go off.  I am quite sure this change over time, so I am not sure what to do with this issue.

kwindrem, 2026-09-14:
This behavior still exists last of v3.73 and v3.80~48.

The thing is that the Multi apparently responds enough so that the GX device instantiates a dbus service for it. I think this is deeded so that the GX device can turn the Multi on.

I think what should be happening is that if the dbus service exists, then /VebusService should get filled in so other consumers of the information can make use of it. What is happening now is that if /Connected is 0, then /VebusService is empty. Much at the system level is based on /VebusService so when it goes empty, the system level stuff assumes there is no Multi.

jhofstee, 2026-09-15:
ok thanks, I will assign it to Jos then.

jhofstee, 2026-09-15:
And for completeness, without a milestone. Contrary to what you might think, turning things on and off isn't that trivial in the VE.Bus world.