victronenergy/venus #1293: Reset Node-RED to factory defaults from dbus / remote console

dirkjanfaber, 2024-06-24:
Instead of the need for a USB stick boot to get Node-RED back to the factory defaults, add a gui option to do that. Which probably means adding an extra option to https://github.com/victronenergy/venus-platform before adding the option to the gui's.

As we don't want users to do this by accident, the user should enter a code for conformation before the script gets called. E.g. the last two digits of the vrm id of the instance or so.

Todo:
- [ ] add Node-RED factory defaults to venus-platform
- [ ] add background script to perform the factory reset. See https://github.com/victronenergy/meta-victronenergy/tree/master/meta-venus/scripts/20-node-red-reset
- [ ] add option to gui-v1
- [ ] add option to gui-v2

Note that we might want to add some more comparable options to this system later. E.g. creating or restoring a backup of your flows.

ReinvdZee, 2024-07-17:
Changes to gui and venus-platform have been pushed to branches. 

Short summary of changes:

- Menu is restructured to have "Node-red" in a submenu. 
- Path `com.victronenergy.platform/Services/NodeRed/FactoryReset` is added. Write the last 4 characters of the uniqueId to it to initiate the reset.
- The gui writes the four last characters to this path when the user confirms he wants to reset Node-red. 
- `venus-platform` handles the reset. Node-red is disabled and everything under `/data/home/nodered/.node-red/node_modules` is deleted. Node-red is left disabled regardless of its state before the reset. 

Screenshots:

[image]

[image]

[image]

@dirkjanfaber fyi.

mpvader, 2024-09-30:
@ReinvdZee can the second check box be ticked?

as in that case, lets just close this issue; having the remaining one in gui-v2 repo only is ok.

dirkjanfaber, 2024-10-01:
Checked second box. Closing this issue as it works as intended.

closed 2024-10-01 by dirkjanfaber