victronenergy/venus #1297: Generator start/stop -> On loss of communication: Add option to start generator when communication with BMS is down?

ReinvdZee, 2024-06-25:
Requested by an installer which experienced a system going down because the communication with the BMS failed. If the generator was automatically started when this happened, the system may not have gone down. 

Currently, the condition "On loss of communication" only starts the generator when vebus communication goes down. Investigate if this condition also works properly when the VE.Can communication goes down.

ReinvdZee, 2024-11-06:
I tested this on 3.51, the "Loss of communication" condition can actually be used for this. This condition evaluates to true whenever any of the other enabled conditions fails to evaluate for 5 minutes. So this situation is covered by enabling a condition which uses the battery service to evaluate.

I had the battery voltage condition set up, and the "On loss of communication" condition set to "Start generator":
[image]

Then, I disconnected the battery monitor and the generator was started 5 minutes later.

I'm not sure if this is clear enough to set up for an end user and if the 5 minute delay should be shorter or configurable.

@mpvader, @philipptrenz any thoughts?

philipptrenz, 2024-11-09:
It's a nice featury I wasn't aware of! Probably only needs to be documented in the manual.

Also, the 5 minutes are fine, I think.

ReinvdZee, 2024-11-11:
@philipptrenz I agree, although it may be a bit strange that it requires a battery condition to be enabled in order for the it to start on loss of communication. And the battery monitor setting must be set correctly. When there is no battery monitor selected, the "loss of communication" condition will also be triggered.

philipptrenz, 2024-11-13:
It would indeed be nice to be able to see which communication is currently watched for. One possibility would be to add a note similar to the following if "On loss of communication" is enabled:

> Based on the active conditions, the genset gets started if one of the following communications fail: Battery, ...

But it would need much logic in the UI or additional data provided by `dbus-generator-starter` and I'm not sure whether the demand for it justifies the effort.

mpvader, 2024-11-13:
No need to do more; this is an existing feature which works fine and is documented.

What could be done, as for many things in the UI, is add an explanatory sentence so people don't have to do look in the manual; part of UX settings redesign or something I'd say.

closed 2024-11-13 by mpvader