victronenergy/venus #1302: v3.40~32: Critical loads box showing, regardless of ESS->grid metering setting

ReinvdZee, 2024-07-03:
Seen on V3.40~32. The Critical loads box is showing in the overview page, regardless of the grid metering setting. V3.32 shows the correct boxes.

See the following screenshots. In both cases, ESS -> Grid metering is set to 'Inverter/Charger' (`/com.victronenergy.settings/Settings/CGwacs/RunWithoutGridMeter == 1`)
v3.32:
[image]

v3.40~32:
[image]

ReinvdZee, 2024-07-03:
Gui-v1 switches between "OverviewHub.qml" and "OverviewGridParallel.qml" based on the presence of the gridmeter. The critical loads box can be shown in "OverviewGridParallel.qml" when the system "has AC out" (`com.victronenergy.settings/Settings/SystemSetup/HasAcOutSystem == 1`). 

The logic of when to show which page has changed [here](https://github.com/victronenergy/gui/commit/8794e5aed69dbeeb3b3dae246bbac4b39d7a5fdb), which uses "OverviewGridParallel.qml" whenever there is a grid meter present. I think this was by design, as per @izak's commit message:

> Additionally, there are some systems that have a grid meter but
are not configured as an ESS system. In such systems we still
want to see an overview of what the entire system is doing,
so we still want to show the GridParallel overview.

>Therefore, if the Grid is measured by a grid meter, switch to
the GridParallel overview instead of using the system type.

The result is that the critical loads box is shown whenever  there is a grid meter (which loads "OverviewGridParallel.qml") and `Settings/SystemSetup/HasAcOutSystem == 1`, regardless of the grid metering setting. The path `Settings/SystemSetup/HasAcOutSystem` is controlled by this setting:

[image]

Which is only visible when "Grid metering" is set to "External". 

Shouldn't the gui switch back to "OverviewHub.qml" if "Grid metering" is set to "inverter/charger", even if there is a grid meter present?

@mpvader @izak

izak, 2024-07-08:
But that is normal. It has always done that if the system type is ESS and a grid meter is present. This logic was never dependent on whether you want ESS to use the grid meter or not.

closed 2024-07-08 by jhofstee