victronenergy/venus #1303: dbus-acsystem: MultiRS devices not combined into acsystem.

ReinvdZee, 2024-07-04:
Found on [this](https://vrm.victronenergy.com/installation/394634/overview) site: the multi RS 3 phase system does not show up in the gui because there is no `com.victronenergy.acsystem...` service. 

`dbus-acsystem` is throwing the following exception:

```
2024-06-27 10:42:28.355007500     raise SignatureBodyMismatchError('Cannot serialize Python type "None"')
2024-06-27 10:42:28.355010500 dbus_next.errors.SignatureBodyMismatchError: Cannot serialize Python type "None"
```

I suppose there is some invalid dbus path that is not expected to be invalid?

ReinvdZee, 2024-07-04:
I turned out the paths `/Settings/Ess/Mode` and `/Settings/Ess/MinimumSocLimit` were invalid on all three multi RS devices. After setting them on one of the Multi RS'es, the AC system service showed up in dbus and on the gui. I don't know why the ESS settings were invalid, though. That may be an issue on the RS side.

martinbosma, 2024-07-05:
Looks like it came from v3.40~20:
[image]

And it was updated to v3.40~34 on 4/7/2024:
[image]

Knowing that `vecan-dbus` does not have timeout on `/Settings/Ess/Mode` and `/Settings/Ess/MinimumSocLimit` and does not invalidate those items by itself, it seems to me that `dbus-acsystem` had copied data around, while `vecan-dbus` did not fetch all items yet and hence the above mentions items were still invalid. Venus v3.40\~20 included `dbus-acsystem` v0.6 and does not have the ["do not copy invalid data" fix](https://github.com/victronenergy/dbus-acsystem/commit/1406959eb566d960bbc76cf61c36da69050fb9f0). This was introduced in `dbus-acsystem` v0.7 in Venus v3.40\~21. So, I believe this issue is an old bug and already solved.

mpvader, 2024-07-05:
Case closed

closed 2024-07-05 by mpvader