victronenergy/venus #1304: npm recipes: better handle optionalDependencies and --omit=dev

mpvader, 2024-07-07:
The NPM definition of an optional dependency is:

> If a dependency can be used, but you would like npm to proceed if it cannot be found or fails to install, then you may put it in the optionalDependencies object.

That conflicts with what we want: reproducible builds + things changing only when we want them to change.

So we need to fix that. 

Secondly, `npm install --optional` is apparently outdated.

So, lets replace that by `npm install --omit=dev --omit=optional`. Which would require:

- [ ] change https://github.com/victronenergy/meta-victronenergy/blob/master/meta-third-party/classes/npm-online-install.bbclass
- [x] add a patch to signalk-server, that changes package.json to make all dependencies non-optional. There was some discussion about the mdns package, but I checked just now and (a) its installed within Venus OS at the moment, and (b) there is this issue: #, and change the rest to be non-optional.
- [ ] check our other recipes.

mpvader, 2025-07-05:
@dirkjanfaber , node-red has one optional package, which is @node-rs/bcrypt. Is that used? Or can we just omit it?
There is another bcrypt installed as well, which is bcryptjs.

I expect that node-red is made such that it uses node-rs/bcrypt when available, and bcryptjs instead.

Since @node-rs/bcrypt is just a binding, to a rust implementation of bcrypt, I'm quite sure it doesn't work at the moment - but would be nice to test / known for sure.

jhofstee, 2026-01-09:
grep says that the only user of "require('bcrypt')" is node_modules/node-red-admin/test/lib/commands/hash_spec.js, which does this:

```
try { bcrypt = require('bcrypt'); }
catch(e) { bcrypt = require('bcryptjs'); }
```

So closing this issue.