victronenergy/venus #1305: add raspberrypi support for Qt6 / gui-v2

jhofstee, 2024-07-08:


WigHlod, 2025-01-25:
Hello, it's look like this issue is mainly related to hardware and is system / kernel settings to the DSI port.
I could try to help and spend some time on it. I run a RPI 3B with the 3.60~15 Beta Firmware for the moment.

The GuiV2 settings is enabled in my VenusOS's settings, but i'm unable to load the actual v2 interface. The RPI keep the legacy one to be display. I assume it's a lock / choice you made to cut the issue temporary for the users.  

Please how can i force it to been able displayed it, face the issues then try to fix it ? 

Regards.

Jimmy062006, 2025-02-26:
I'd like to see/help with this as well.

mpvader, 2025-04-16:
Hi both @WigHlod & @Jimmy062006 ; sorry for the slow replies. I don't see notifications / there are too many.

To get going with this you can build gui-v2 straight from its sources, using gui-v2 instructions in its repo: https://github.com/victronenergy/gui-v2/wiki/How-to-build-venus-gui-v2.

And then copy that binary onto the pi, and try to run it.

The code for starting it under normal circumstances is this one: https://github.com/victronenergy/meta-victronenergy/blob/master/meta-ve-software/recipes-ve/gui-v2.bb#L31

Note that to get going on this, you'll need to know how to build Venus OS from source, get involved in OE work, etcetera etcetera.

And, make sure once you have a plan to check in here if its in line with what the maintainers have in mind for thos.

If you are really up for that and up to speed with all that - then great.

Note that in case someone was thinking about it: there is a run time flag, or environmental variable to make gui-v2 run without hardware acceleration; which seemed like a good solution - but we tried that already and doesn't work; somehow makes one or two CPU cores on the Pi be 100% busy.

jenskaschub, 2025-09-07:
Waiting for V2 on raspberry-displays too.
But for me it would be enough having access to the nodered dashboard as a page on V1.
Possible?

Born2Ski, 2026-11-11:
@mpvader @jhofstee checking in as suggested before going further. I looked into what blocks hw acceleration on the Pi 2 / 3 / Zero 2 (`raspberrypi2` machine) and have a branch on top of v3.85. Before turning it into a PR against master, I'd like to know if this is in line with what you have in mind.

**What I found**

The kernel side is fine: the 5.10.110 rpi kernel already builds `DRM_VC4`, the 7" panel/bridge and `edt-ft5x06` as modules. Three other things keep gui-v2 off the GPU:

1. `config.txt` doesn't load `vc4-kms-v3d`, and the dtbo isn't deployed to the boot partition, so vc4 never probes.
2. mesa is built without the `vc4` gallium driver (only etnaviv/kmsro/lima/virgl/swrast).
3. `mesa-megadriver` is a BAD_RECOMMENDATION unless `VENUS_GPU = "1"`, which only sunxi sets.

On top of that, qtplatform uses `linuxfb` on all Pis, so gui-v2 renders in software, which matches the CPU load you saw. Point 2 may also explain why a newer kernel alone didn't help. The Pi 5 is in the same situation: no `v3d` in mesa, and it uses linuxfb.

- `raspberrypi2`: deploy `vc4-kms-v3d` + `vc4-kms-dsi-7inch`, set `dtoverlay=vc4-kms-v3d` and `display_auto_detect=1`, set `VENUS_GPU = "1"` and `START_GUI = "start-gui"` (like the Pi 5)
- mesa: add `vc4` unconditionally, since mesa isn't machine specific
- qtplatform for raspberrypi2: `eglfs_kms` on `/dev/dri/by-path/platform-soc:gpu-card`, falling back to linuxfb when the KMS device is missing. swupdate doesn't touch the boot partition, so existing installs stay on the firmware fb.
- display-hotplug (separate commit): it assumed one connector per card, but vc4 also has Writeback/DSI connectors, so pick the one with a ddc bus

**Questions**

1. Is full KMS by default on Pi 2/3 acceptable? Legacy `hdmi_*` / `lcd_rotate` settings stop working, and the official 7" touch moves from the firmware driver to edt-ft5x06. Or would you prefer it opt-in?
2. OK to add `vc4` to the shared mesa build? It makes mesa-megadriver slightly bigger on sunxi as well.
3. Is the linuxfb fallback enough for existing installs, or do you want a way to update `config.txt` / overlays on the FAT partition?
4. `START_GUI = "start-gui"` needs the private start-gui package built for raspberrypi2. Is that fine on your side?
5. Pi 4/5 would need `v3d` (+ kmsro) and their own overlays. Keep that as a follow-up?

Born2Ski, 2026-11-14:
Follow-up to my previous comment: gui-v2 now runs hardware accelerated on real hardware, a Raspberry Pi 2 Model B with a 1080p HDMI display on Venus OS v3.85.

```
qt.rhi.general: OpenGL VENDOR: Broadcom RENDERER: VC4 V3D 2.1 VERSION: OpenGL ES 2.0 Mesa 24.0.7
```

venus-gui-v2 runs on `eglfs_kms` with the vc4 driver loaded and stays up. Plugging the HDMI display in after boot also works: display-hotplug restarts start-gui, which then picks gui-v2.

**How I tested**

Not with a full OE build yet. To test on the stock v3.85 image I put the same changes as my [branch](https://github.com/Born2Ski/meta-victronenergy/tree/rpi3-gui-v2-v3.85) into a SetupHelper package, [RpiGuiV2](https://github.com/Born2Ski/RpiGuiV2):
- the `config.txt` lines and overlays
- a `vc4_dri.so` cross-built from mesa 24.0.7 with the OE patches of b3.80
- the qtplatform / `qt-kms.conf` and display-hotplug changes
- gui-v2 from the release feed, and a start-gui switcher

**Found along the way**

1. mesa bug: `init_matrix_stack()` and `_mesa_PushMatrix()` allocate `GLmatrix`, which has `alignas(16)` members, with `calloc()` / `realloc()`. On 32-bit ARM that gives only 8-byte alignment. My driver was built with clang, which uses `vst1.64 [rN:128]` stores there, so gui-v2 died with SIGBUS (alignment exception) as soon as it created its GL context. Victron's GCC-built mesa runs this code fine on 32-bit ARM, so an OE build is probably not affected, but the allocation is wrong either way. Fix: [0001-mesa-align-the-matrix-stack.patch](https://github.com/Born2Ski/RpiGuiV2/blob/main/build/patches/0001-mesa-align-the-matrix-stack.patch)
2. Overscan: with KMS the firmware passes its default overscan to the kernel as 48 px margins (`video=HDMI-A-1:1920x1080M@60,margin_left=48,...`), which shows as a black border. `disable_overscan=1` should remove it; once I've confirmed that, I'll add it to the branch.
3. CMA: with the 256 MB CMA from the vc4 overlay, `CmaFree` dropped to about 1 MB while gui-v2 was running at 1080p. That may just be movable pages borrowing CMA, but I'm keeping an eye on it.

**Not tested yet:** Pi 3 / Zero 2 W, the official 7" DSI display, and an OE build of the branch.

The questions from my previous comment still stand.

Born2Ski, 2026-11-15:
GPU is being used instead of CPU

[image]

Feel free to test it, I only have access to RPi2 connected to HDMI at the moment. 
https://github.com/Born2Ski/RpiGuiV2