victronenergy/venus #1307: Generator menu: Hide quiet hours menu items when quiet hours is disabled.

ReinvdZee, 2024-07-09:
Add small change in the gui to hide the quiet hours menu items when the quiet hours setting is disabled. Makes the UI cleaner. 

And since its for 3.50 and a nice-to-have, this will only be implemented in gui-v2.

kwindrem, 2024-07-09:
Not good. You need to allow the quiet time to be set before activating it. A previously set quiet period could activate while making changes. 

I also dislike disappearing configuration items. It leads to "where was that settign? it was there a minute ago"

philipptrenz, 2024-07-18:
That's a good point, @kwindrem. 

For restructuring the settings menu, we're planing to have sub headlines to divide long lists of items. This could also be a nice solution here to just visually separate the quiet hours related items from the rest instead of hiding them, which should still make it less overwhelming and easier to grasp.
[image]

ReinvdZee, 2025-02-04:
The menu for quiet hours will be completely revised. This is not relevant anymore. Closing

closed 2025-02-04 by ReinvdZee