victronenergy/venus #1314: signalk: default canbus interface for Cerbo MK2 is wrong

mpvader, 2024-07-16:
For the Cerbo MK2, (and maybe other models as well)

What needs to be done is 

1. correct this one, perhaps even fix it automatically for people already running into that issue and most likely just giving up on it.
2. check for all models (cerbo old, cerbomk2, cerbo-s, ekrano) and correct where necessary.

Slack: https://victrondevelopment.slack.com/archives/C7C1ZU6UD/p1721065005510819

mpvader, 2024-07-28:
- Cerbo GX: can0 and can1
- Cerbo-S GX: can0
- Cerbo GX MK2: vecan0 and vecan1
- Ekrano GX: vecan0 and vecan1
- MultiPlus-II A13: can0

mpvader, 2025-05-24:
This is not trivial to fix as the signalk config is now based on the machine; and cerbo gx and cerbo gx mk2 are the same machine.

What I’ll do is change the default to match the cerbo gx mk2, as that is now the only model still sold.

mpvader, 2025-05-24:
https://www.victronenergy.com/live/venus-os:large#can-bus_ports

itemir, 2025-11-08:
We were discussing this on Discord, but I wanted to chime in here as well. This is becoming a more common issue as new users have Cerbo GX MK2 and Ekrano GX units. It breaks the out-of-the-box experience and discourages many first-time users.

> What I’ll do is change the default to match the Cerbo GX MK2, as that is now the only model still sold.

I think this will address the problem for new users going forward. Any plans to implement it?

jhofstee, 2025-11-20:
Mentioned patch should set a reasonable default for the NMEA 2000 interface. It adjust the settings if the configured CAN-bus interface doesn't exists and a NMEA 2000 candidate is found. It can't hurt if signalk users actually test this...