victronenergy/venus #1316: Remove/fix the gui option to clear genset errors. See slack thread

ReinvdZee, 2024-07-16:


ReinvdZee, 2024-07-18:
As discussed in the slack thread, `dbus-modbus-client` will create the path `/ClearError` if the genset supports clearing error remotely. See https://github.com/victronenergy/venus/issues/1312

philipptrenz, 2024-07-18:
As generator warnings and errors are shown in the gui-v2 notification center and can be acknowledged there to be moved into the history, we should hook into this user input to trigger the `/ClearError` path rather than introducing a new button in the generator device list item. Thereby, they stay in sync with the controller, if clearing is supported.

philipptrenz, 2024-07-19:
As discussed with @ReinvdZee and @martinbosma, this might need a bit more of research to fully understand what clearing or acknowledging errors means for different genset controllers.

closed 2024-09-23 by philipptrenz