victronenergy/venus #1324: dbus-modbus-client: Not detecting DSE genset

ReinvdZee, 2024-07-25:
On Venus 3.50~5, dbus-modbus-client seems to fail to detect the DSE genset. On 3.40, it works fine. I have manually added the IP of the genset (gateway in this case). A scan also does not show the genset. A lot of systems in Groningen can find the genset usually. I used [this one](https://vrm.victronenergy.com/installation/376477/overview) to test this. It is not used for anything else at the moment. 

Logs:
```
3.40:

2024-07-25 12:05:10.251571500 *** starting dbus-modbus-client ***
2024-07-25 12:05:16.966145500 INFO     Waiting for localsettings
2024-07-25 12:05:17.223533500 INFO     registered ourselves on D-Bus as com.victronenergy.modbusclient.tcp
2024-07-25 12:05:17.333359500 INFO     [tcp:10.230.1.59:502:1] Found DSE 7310 MKII
2024-07-25 12:05:17.360987500 INFO     [tcp:10.230.1.59:502:1] DSE status code register is not available
2024-07-25 12:05:18.783740500 INFO     registered ourselves on D-Bus as com.victronenergy.genset.dse_4294967295
2024-07-25 12:05:18.998162500 INFO     [tcp:10.230.1.59:502:1] Detecting engine status by RPM
2024-07-25 12:05:19.165522500 INFO     registered ourselves on D-Bus as com.victronenergy.tank.dse_4294967295_0

3.50~5:

2024-07-25 12:07:01.744362500 *** CCGX booted (0) ***
2024-07-25 12:07:02.169069500 *** starting dbus-modbus-client ***
2024-07-25 12:07:08.914829500 INFO     Waiting for localsettings
2024-07-25 12:07:09.201557500 INFO     registered ourselves on D-Bus as com.victronenergy.modbusclient.tcp
2024-07-25 12:09:41.644932500 INFO     Starting background scan
2024-07-25 12:09:41.672488500 INFO     Scanning 10.230.0.0/22
2024-07-25 12:11:15.408652500 INFO     Scan completed in 93 seconds
```

@mansr, @philipptrenz

philipptrenz, 2024-07-25:
Hm, works fine at my network configuration:
```
@4000000066a2476f15dc8ccc *** starting dbus-modbus-client ***
@4000000066a247710496b884 INFO     Waiting for localsettings
@4000000066a2477108452e0c INFO     registered ourselves on D-Bus as com.victronenergy.modbusclient.tcp
@4000000066a247730508bc04 INFO     Starting background scan
@4000000066a2477306bc20f4 INFO     Scanning 192.168.179.0/24, 192.168.254.0/24
@4000000066a247761604e4ec INFO     [udp:192.168.254.104:502:1] Found Energy meter: Victron Energy VM-3P75CT
@4000000066a2478c2f471894 INFO     [tcp:192.168.254.20:502:1] Found Generator controller: Deep Sea Electronics 4620/4623
@4000000066a2478c36cff1bc INFO     registered ourselves on D-Bus as com.victronenergy.genset.dse_65537
@4000000066a2478d073af99c INFO     registered ourselves on D-Bus as com.victronenergy.tank.dse_65537_0
@4000000066a2479405349e64 INFO     [udp:192.168.254.104:502:1] Found Energy meter: Victron Energy VM-3P75CT
@4000000066a247a22e47501c INFO     Scan completed in 47 seconds
```

I would guess this is related to https://github.com/victronenergy/dbus-modbus-client/commit/822eb6c460d103089d98b3bb48bc87be28174d73

mansr, 2024-07-29:
The probe order changed, and the Comap probe (which was previously after DSE) triggered an exception that aborted the whole thing. Fixed (https://github.com/victronenergy/dbus-modbus-client/commit/9acfa987ff29165b71c959d64a6dbe806cdf62e5) by ignoring such exceptions.

closed 2024-07-29 by jhofstee