victronenergy/venus #1325: vebus updater: Fix error 123 when updating MultiPlus-II 48/3000/35-32 (2699xxx)

vejpasop, 2024-07-25:
Error 123 description:  "firmware file supplied does not match the connected device model" 
Affected model: MultiPlus-II 48/3000/35-32 (2699xxx)
Bug since all Venus versions up to date.

VictronConnect is also affected. Currently only VeFlash can be used to update a 2699xxx

When trying to update a single Multi or a system consisting of solely Multi models 2699xxx the above error will prevent the Multi's from being updated. 
In a combined system, parallel or multi phase, when one or more units is a 2629xx the update will commence properly. 

### Reason:
Before switching to bootloader mode the updater reads the compatibility product id, in this case 2629xxx instead of the actual product id (2699xx). Which compared to the expected product id 2699 from the supplied firmware file resulting in error 123

### Todo
- [ ] Fix in vbdup

vejpasop, 2024-07-26:
Fixed in vbdup v1.13

mpvader, 2024-07-29:
Hi @vejpasop , what is the error message that the customer would see when trying to update such unit?

Error 123?

Or vbdup-123?