victronenergy/venus #1327: vebus: Enable PV inverter on AC output during wrong phase rotation and while not all AC inputs are powered

vejpasop, 2024-07-26:
In systems with PV inverters running on the AC output of a Multi/Quattro the ESS/PV inverter assistant switches the PV inverters off by raising the AC output frequency during a short time while attempting to switch to grid. This in order to prevent the batteries from being overcharged.

When switching to grid fails, due to a phase rotation issue or due to missing AC input voltage, while at least on Multi/Quattro detects AC input, the PV inverters remain disabled. 

Possibly we can fix this by letting mk2-dbus set the ignore AC input in this case.  
The wrong phase rotation detection takes 30 seconds to become valid. During this time the PV inverters on the output will be disabled.

We need to consider any condition that would not allow connection to the grid. Such as high frequency, phase variations. etc

mpvader, 2024-07-27:
Hey Jos, isn’t it better to first explore the options of fixing this in vebus fw?

mpvader, 2024-12-05:
https://victrondevelopment.slack.com/archives/C04RYU0UMDE/p1720947326049649