victronenergy/venus #1328: Add option "system has normal AC loads"; allowing installer to set that to No.

mpvader, 2024-07-27:
And then on each UI, that box should either be hidden; or if that is complex, then just show 0. My proposal:

Add the new setting, which obviously defaults to yes.

- [ ] systemcalc: make those paths return 0 when its set to 0.
- [ ] gui-v1: hide if trivial, otherwise do nothing since it will automatically say 0 which is also an improvement.
- [ ] gui-v2: hide if trivial, otherwise do nothing since it will automatically say 0 which is also an improvement.
- [ ] VRM Portal: . Since current dashboard is EOL, do nothing and it will say 0
- [ ] VRM Widgets: hide it. Effort is worth it.
- [ ] VRM Watch App: hide id. Effort is worth it.
- [ ] HTML5: either way is OK
- [ ] EVCS UI: either way is OK 

Background of this request is systems in Austria, and possibly elsewhere in the world, where all loads are behind the ESS system, for being backupped. But part of the installed (AC-coupled) PV is not, due to 1:1 rule. And as its now, it technically works but all those UIs all the time show some measurements for AC loads, while there are no AC loads; what is shown is just timing and measurement differences.

[Slack](https://victrondevelopment.slack.com/archives/C04RYU0UMDE/p1721926219943959)

A visual, but note that this won't happen soon since we're not fixing it this way on VRM, read above.

[image]

mpvader, 2024-07-27:
@izak pls let me know what you think of above approach; required effort in systemcalc; and make sure to do a quick test (!)

izak, 2024-07-30:
I think we need to integrate this with the existing setting, `/Settings/SystemSetup/HasAcOutSystem`. This already hides AC-out for systems that only has loads on the input side. Perhaps this needs to be an option list where you can choose between the three options: Only input side, only output side, or both.

A crude "make it zero if it isn't invalid" proof of concept reveals:

Before:
[image]

[image]

After:

[image]

[image]

izak, 2024-07-30:
Additionally, an experiment where instead of setting the consumption to zero, it is set to invalid:

GUI v1:

[image]

VRM:

[image]

izak, 2024-09-13:
# Screenshots for GUI-v1 proposal.

New setting for specifying position of AC loads. The case for systems with no grid meter is a bit unclear perhaps. I don't want to hide the setting if there is no grid meter, because that makes it impossible to reach the setting when there is a grid outage.

## Options

[image]

[image]

## Some toast text to tell the user what is going on:

[image]

[image]

## Behaviour of overview

In both examples here, there is a PV-inverter on the input and some load on the output.

With setting in the "normal" position:

[image]

With setting in the "output only" position:

[image]

## VRM
With setting in the "normal" position:

[image]

With setting in "output only" position:

[image]