victronenergy/venus #1330: Trigger GX alarm relay also on Lynx Smart BMS warnings

ReinvdZee, 2024-07-30:
Currently, the GX alarm relay only triggers on alarms. Should there be an exception for Lynx Smart BMSes (and possibly other devices as well) to trigger the alarm relay on warnings? The purpose of this is to notify the user of low cell voltage before the system shuts down.

And do we then trigger on all warnings? Or only on low cell voltage?

@mpvader @vejpasop

mpvader, 2024-07-30:
All warnings of the lynx smart bms & NG. And relay on the gx needs to be triggered as well.

and during implementation pls check what effect this has on mgd batteries like Pylontech

msigman, 2024-07-30:
Super happy to see this @ReinvdZee, this is something I have been missing and will be greatly appreciated.

I suspect users would appreciate having a toggle in Settings to enable/disable this feature. If you wanted to get fancy you could make it a dropdown, so the user can choose which relay to trigger for them (None/Relay1/Relay2).

I am happy to help test this if/when needed.

ReinvdZee, 2025-01-06:
I think the desired behavior is as follows:

1. Engage the alarm relay when there are one or more unacknowledged warnings triggered by a Lynx Smart BMS (prod ID's 0xA3E4 - 0xA3E5)
2. Disengage the relay when the notification is acknowledged or the warning condition is resolved.
3. Do not engage the buzzer
4. Show the notification as a warning.

Since these warnings should not engage the buzzer, we cannot treat it as an alarm. Although its possible to implement an exception for Lynx BMS warnings silently, maybe its useful to update the venus-platform API with a field which explains why the relay is engaged while there are no alarms but only warnings.

Something like `/com.victronenergy.platform/Notification/LynxWarning` which will only be 1 when there are unacknowledged warnings triggered by the Lynx Smart BMS. We can also add a new notification type for this (since the behavior is different from the available types), but I don't think that's necessary.

msigman, 2025-01-29:
> I think the desired behavior is as follows:
>
>
>
> 1. Engage the alarm relay when there are one or more unacknowledged warnings triggered by a Lynx Smart BMS (prod ID's 0xA3E4 - 0xA3E5)
>
> 2. Disengage [...]

This would be a significant improvement. 

But I have to ask myself, why wouldn't we want the buzzer to sound on these? And that makes me think, perhaps the real issue here is that Lynx BMS Low Cell Voltage warnings (and similar ones) should actually be alarms? Then we wouldn't need this custom logic. 

Either option would be a significant upgrade from the current rather dangerous setup though. 

Let me know how I can help.

msigman, 2025-09-09:
@mpvader Thanks for the note. I appreciate and understand the recommendation. Frankly the only reason that's not ideal for my setup is that I'm already using the Lynx relay for alternator control. If there were two relays or I could designate a GX or GX Extender relay for this I would gladly do so.