victronenergy/venus #1335: Move relay-controlled generator settings into device list

philipptrenz, 2024-08-20:
As planned in the GUI-v2 settings redesign project, all devices should appear in the device list. For gensets, this is already the case for connected ones. But the pages for relay-controlled ones are still located in the settings as "Generator start/stop".

[image]

Additionally, there is a function to track the generator state via a digital input using the "Generator" role. By this, a "Generator" will show up in the device list.

[image]

To harmonize connected and relay-controlled gensets:

- Also the relay-controlled one should be displayed in the device list as soon as relay 1 is configured to "Generator start/stop" and/or one digital input is configured to "Generator"
- The device should be merged together with the one from the digital input "Generator" option to accurately display also the state within the same device representation

As it is planned to use the digital input for improving accumulated run time counters for relay-controlled gensets (see https://github.com/victronenergy/venus/issues/337), these two issues should be synced to provide the best user experience.

philipptrenz, 2024-08-20:
@mpvader any objections?

mpvader, 2024-09-25:
No objections; but, for gui-v2 pls have a good think how valuable it is there to have it in the device list, since the whole concept of device list is quite of less value than it is in gui-v1

ReinvdZee, 2024-10-03:
We won't do this now, so this issue can be closed.

closed 2024-10-03 by ReinvdZee