victronenergy/venus #1342: Explore what is needed to (for a while) offer users the choice between gui-v1 and gui-v2

mpvader, 2024-09-05:
1) default to gui-v2 on Cerbo GX and Ekrano GX
2) reword the texts in gui-v1 and gui-v2, VRM, and urls, includes removing the word beta
3) fix things in VRM
4) in gui-v1, show a toast before stopping it.
5) make some onboarding page? like - here is are new gui, scan this qr code to read the blog about it. and also somewhere note that in case there are issues, there is an option, for now, to still use the old one, and here is where to find that.

mpvader, 2024-09-16:
[image]

mpvader, 2024-09-16:
and this is how it now looks in gui-v1

[image]

jepefe, 2024-10-08:
This is the proposal.

The pop-up shows right after the gui is loaded and asks the user to switch now to the new gui, then there are three possible scenarios
  1. Skip and do not make a choice -> The pop-up keeps showing on every gui restart (power cycle/fw update..)
  2.  Opt to do not switch to gui-v2 yet -> The pop-up is not shown anymore.
  3.  Decide to switch to gui-v2

For scenario #2, a new setting holds if the user has made a choice. We can use this setting on a future release to force the pop-up to be displayed again.

@mpvader Any comments? If you agree I will push the branch and add it to the todo.

https://github.com/user-attachments/assets/743041ee-2b2c-4c3f-97c6-824e2994e2f3

[image]
[image]
[image]

jepefe, 2024-10-09:
Applied changes suggested by @mpvader 
Pushed an added to the ToDo.

[image]
[image]

mpvader, 2024-10-09:
nice @jepefe . Jeroen will include it in the build

mpvader, 2024-10-23:
all done

closed 2024-10-23 by mpvader