victronenergy/venus #1345: Lynx parallel BMS: Do not aggregate BMSes with different nominal voltage levels.

ReinvdZee, 2024-09-16:
The Lynx smart BMS paralleling feature automatically aggregates ALL Lynx BMSes with the same N2K device instance. This also means that BMSes managing totally different battery banks will be aggregated if they have the same N2K device instance. This leads to confusion when updating Venus OS from a version without the feature to a version with the feature, as two independent BMSes on different voltage levels will suddenly be aggregated. 

The Lynx parallel BMS feature should be changed such that BMSes with different (nominal) voltage levels are not aggregated, even if they have the same N2K instance.

nmbath, 2024-09-16:
While I think this is a great idea. I think it will lead to further challenges and issues.

 It is working around how a system should be configured and that will then give further issues. I have a system that has a 12v and 24v (*2) BMS devices. It is annoying, but the configuration of the device prioriyu is easy.

Consideration should be taken into what else would fail and other challenges integrating an environment. It would be better to just pop a message advising the user configures the system correctly and changes on of the BMS devices to a different data instance.

ReinvdZee, 2024-09-17:
@nmbath, Would there be any case in which you'd want to aggregate BMSes of different nominal voltage levels? What challenges and issues do you see here?

nmbath, 2024-09-17:
@BJansink in my view there is a broader issue. The system is really designed around a single voltage. So from a DVCC perspective you can only control one battery voltage. From a display in seeing what is happening is really designed for single voltage. Therefore, I would expect users with multi bank voltages to go down a route of a single VenusOS device per voltage. This is what I have done.

If a single VenusOS device is to be considered then at a minimum the DVCC needs to operate on all voltages being monitored.

As an example I have two OrionXS devices charging my 12v bank. Using a single VenusOS I can't control them as well as the devices on the 24v bank.

The UI component is a second order issue as does allow multiple devices to be seen and monitored and the overview screen becomes a what is the system as a whole doing. However, once you go to two VenusOS devices I any VE.Can connected devices (Such as CAN connected MPPTs and NMEA 2k from the other VenusOS device) a mass of extra devices materialise and confusion rains.

As an example my 12v VenusOS devices sees the 24v MultiPlus from N2K out on the other Cerbo and wants to start controlling it !

A lot of thought is needed on the design and operation of multi voltage systems. I hope this has given some useful insights.

ReinvdZee, 2024-09-17:
I understand that controlling two different battery banks with DVCC is currently not possible. Nevertheless, I'm told that there are quite some systems which have e.g. a main 24V bank (controlled by DVCC) and a secondary 12V bank for small appliances. Currently, when users update such a system from an older venus OS version to a version with the paralleling feature, the unrelated battery banks are suddenly combined into one. 

AFAIK, there are no cases in which you'd want to combine banks of different voltage levels into 1, so we might as well prevent that. @nmbath I agree that we should think about how to handle different battery banks with respect to DVCC, but that is a much broader issue than this.

closed 2024-09-19 by jhofstee

nmbath, 2024-09-19:
@ReinvdZee I agree there is no situation where you would want to aggregate two BMS of different system voltages. However the solution to this is change the device instance, as documented. Working around that key design/configuration rule, will just mean that people do not change the device instance and what other issues will that create. Take external devices that work on the basis of different network address, they will then break, and they now need to add more logic to validate if two BMS are working together. It almost makes the device instance must be the same to aggregate a waste of time, as three BMS's devices, all with the same device instance and two at 24v and one at 12v will mean that the two 24v units are aggregated and teh 12v not. Set the 12v to a different devcieinstance and it would never have been considered in the first place.

Anyway I see the change has been made so it is a moot point. We will need absolute clarity on what the configuration guidelines are, and ensure that alternator controllers (the known one today) all follow the same rules of what can and can not be aggregated.

mpvader, 2024-09-23:
@nmbath : there are existing boats out there; with both a 24V and a 48V battery bank. And device instance both to 0, because there never was a need to change that.

And what the feature as implemented did now was break the system, make it work in an unwanted way, after a simple Venus OS update. Which is not wanted.

Non technical end users of boats should be able to update Venus OS without running into trouble, calling their yard etc.

With above change that issue is fixed.

Most likely I'll take this in to a Venus OS v3.4x backport.