victronenergy/venus #1351: Add warning when accessing html5 app on http:// while security profile is set to secure

mpvader, 2024-09-30:
it should say that to access this page, user needs to change Settings -> General -> Access to unsecure.

mman, 2024-09-30:
@mpvader Just as a side note... I have made the necessary underlying changes in the app itself to work correctly in either case, so even if you specify `Secure Profile` `Weak` or `Secure` the `/app` will actually work just fine as of the latest release in Firefox, Chrome, or any other https capable device that is able to enter the Venus password one time... on the MFD it does not make sense as we do not have a keyboard to enter the password, but elsewhere it already works...

mpvader, 2024-09-30:
understood, nice that that works always @mman .

Still though, when accessed from an MFD, I'd like to give the user a tip on how to make it work.

Jeroen already has an option in mind I think, by serving a page on http://[IP]/app and https://[IP]/app.

Lets leave this to Jeroen for now.

mpvader, 2024-10-04:
Nice!  One change in the last text

```
To make the Victron app work on such a display, change this setting on the Victron GX device (and not on this MFD):
(New line)
Settings -> …
```

mpvader, 2024-10-18:
tested, works nicely

[image]

closed 2024-10-18 by mpvader