victronenergy/venus #1354: SoC don't uses decimal digits given from BMS via Can

patrick-dmxc, 2024-10-02:
Hi, in my setup i use Batrium, its sending the SoC with 2 decimal digits to the Cerbo. but it not apears in VRM or Node-Red with this resolution.
The Bartiom uses the Victron-Protocoll

In a nother Setup i have a Lynx-Shunt, there the SoC is with multiple decimal Digits in Node-Red and one decimal digit in the VRM.
Can someone fix this?

mpvader, 2024-11-06:
Hi @patrick-dmxc , this is quite hard to fix, due to the way our system and architecture is set up. And having two decimal SOC is quite rare: I'm afraid this is not going to be looked at anytime soon

closed 2024-11-06 by mpvader

patrick-dmxc, 2024-11-06:
I was thinking, this is not as complicadet. Because with a lynx Shut its possible