victronenergy/venus #1355: Possible add n2k Tank sensor no name ??

carlosgaros, 2024-10-03:
[image]

mpvader, 2024-10-04:
Hi,

could you ask this on https://community.victronenergy.com instead?

thank you!

closed 2024-10-04 by mpvader

carlosgaros, 2024-10-04:
I can ask but no one will add supported device

carlosgaros, 2024-10-04:
Maybe you need to add classes 80/150 for it to work ?
[image]

carlosgaros, 2024-10-06:
> Hi,
> 
> could you ask this on https://community.victronenergy.com instead?
> 
> thank you!

https://community.victronenergy.com/t/nmea2000-tank-sensor-cx5001-not-shown/6402

carlosgaros, 2024-10-12:
done! Per v3.50~31