victronenergy/venus #1356: AC system service makes mqtt-rpc freeze while getting devices list

jepefe, 2024-10-03:
The issue becomes noticeable when a VictronConnect user connects to an installation and the connection progress gets stuck at 40%(waiting for the device list from mqtt-rpc) for about 30 seconds.

At that point to reply with the devices list, mqtt-rpc iterates all the DBus services and performs a `GetValue` on `/Devices` and acts accordingly to the result of the call, the problem arises when processing the `com.victronenergy.acsystem.socketcan_can0_sys0` service then the execution freezes for about 30 seconds (DBus timeout). After that, the exception is caught and the loop continues normally.

I noticed that GetValue is not implemented for parent nodes such as `/` and `/Devices`, which might be related. See below.

Seen on v3.50 ~27 & ~28

```
root@einstein:/opt/victronenergy/mqtt-rpc# dbus -y com.victronenergy.acsystem.socketcan_can0_sys0 / GetValue
[]
root@einstein:/opt/victronenergy/mqtt-rpc# dbus -y com.victronenergy.acsystem.socketcan_can0_sys0 /Devices GetValue
Traceback (most recent call last):
  File "/usr/bin/dbus", line 374, in <module>
    ret = obj.object.get_dbus_method(method.name, iface.name)(*args)
AttributeError: 'NoneType' object has no attribute 'name'
```

mpvader, 2024-10-04:
How urgent is this @jepefe ?

Affects all systems with an RS?

mpvader, 2024-10-04:
It affects all systems with an RS: so super urgent since this is a blocker for v3.50 release; @izak

izak, 2024-10-07:
As expected, this is caused by the move to dbus_fast from dbus_next, which was patched to deal with a null interface on the call.

The fix is just to make mqtt-rpc call it with the right interface instead of null.

```
method call time=1728296705.510618 sender=:1.134 -> destination=:1.105 serial=5 path=/Devices; interface=(null); member=GetValue
```

closed 2024-10-09 by jhofstee