victronenergy/venus #1358: Battery CCL ignored by MPPT control

weichenb, 2024-10-06:
CCL defined by BMS or in DVCC should actually assure battery health.
Such definitions are properly instructed to Multiplus's, but not to MPPTs.
When CCL is set to 0 by BMS MPPTs continue charinging until having reached float voltage and are NOT properly controlled by VenusOS.
In the example below please see a case where a BYD LVL provided CCL=0, but MPPT provided up to 46A during a period of 6 minutes.
[image]

I would appreciate, if MPPTs are not only controlled by CVL but also by CCL.

thepaulcooper, 2024-10-06:
I wonder if this is why I get very frequent "**BYD Premium LV battery - Warning - High voltage**" messages where the reported voltage momentarily peaks at over 56 volts?  Weichenb, do you get the same?  My setup is a MultiPlus II GX, MPPT RS 450/100 and 12kWh of BYD LV batteries with the BYD BMS.

weichenb, 2024-10-06:
@thepaulcooper : On my side I haven't experienced any over-voltage issues.
Setup:
- MultiPlus-II 48/5000/70-48 - 3 phase setup
- SmartSolar MPPT VE.Can 250/100 rev2
- Fronius Symo 6.0-3-M attached to AC-OUT

asdt1803, 2024-10-06:
Is it possible, that this issue not only affects BYD batteries? I'm having similar issues with my Gobel Power batteries. 
The Pace BMS also does not feature a function to limit CCL. As the DVCC CCL does not have any affect on my RS450/200 and RS450/100 MPPT, it happens on days with highly fluctuating Solar Power, that one of the batteries goes into Cell overvoltage protection when the weakest cell of the batterypack peaks due to a suddenly high charge current. 
Unfortunately the Pace BMS transmits only the lowest battery voltage over CAN. As the voltage of the battery in protection state decreases, the Victron system continues to charge the remaining batteries. 
But: It does that without respecting the CVL-Limit configured in the DVCC setting and in the BMS settings (in my case both configured to 55,2 V)! I can measure up to 57,6 V directly at the Lynx Busbar, so that the remaining batteries enter into overvoltage protection one after another until all batteries are in protection state.
[image]
 
The DMM on the picture is connected directly to the Lynx Busbar.

Strangely the overshooting voltage is not visible in the VRM graphics even though this bevaviour sometimes occurs for more than a minute.

My System:

3 x MPII 8000
1 x RS450/200
1 x RS450/100
1 x Fronius Symo 10.0-3 in AC input
4 x Gobel Power PG-SR1-PC200 (Pace BMS)
29,92 kWp

weichenb, 2024-10-06:
> My System:
> 
> 3 x MPII 8000 1 x RS450/200 1 x RS450/100 1 x Fronius Symo 10.0-3 in AC input 4 x Gobel Power PG-SR1-PC200 (Pace BMS) 29,92 kWp

As I can see and think to understand, I suppose that your PACE BMS is not supported by dvcc.py
```
	0xB004: _lg_quirk,
	0xB009: _pylontech_quirk(),
	0xB00A: _byd_quirk,
	0xB015: _byd_quirk,
	0xB019: _byd_quirk,
	0xB029: _pylontech_pelio_quirk,
	0xA3E4: _lynx_smart_bms_quirk,
	0xA3E5: _lynx_smart_bms_quirk,
	0xA3E6: _lynx_smart_bms_quirk,
	0xA3E7: _lynx_smart_bms_quirk,
```
In the moment only LG, BYD, Pylontech and Victron BMS seem to be supported by that script.

asdt1803, 2024-10-07:
> Btw. what voltages did you set in Pace BMS?
> 
> In case you configured Pylontech simulation in Pace, voltage is directly taken from Pace.

The BMS uses Pylontech protocal. In the BMS-settings I left the default value of 55.2 V. 
In fact I usually set a lower voltage via CVL, especially when I'm away from home, to avoid this charge voltage overshoots.  As said, CVL usually works for me. The only trouble occurs with full or nearly full batteries in combination with highly fluctuating solar power as described above. In that case the Charge voltage exceeds the configured limits for whatever reason.