victronenergy/venus #1370: Generator start stop: Warm-up skipped in some cases

ReinvdZee, 2024-10-15:
With the improvements made [here](https://github.com/victronenergy/dbus_generator/commit/fe7beb3af842ddb74aec067569cb87ad88232590), we actually introduced an issue. Startstop will only be in the warm-up phase very briefly (~1 second) when the genset does not report fast enough that it is not stopped anymore (warm-up, running, etc.).

ReinvdZee, 2024-10-15:
@philipptrenz and I agreed to fix this by extending the warm-up phase until the generator is reported "not stopped" anymore. So the warm-up phase will actually start when the genset is no longer in the stopped state. This is actually an improvement, since the generator is intended to run during the warm-up phase. 

For relay controlled gensets with a digital input used for runtime counting, the same applies. The few seconds it takes the generator to be started will not be counted as 'warming up'. The warm-up phase starts as soon as the digital input closes.

For relay controlled gensets without a digital input, nothing changes and the warmup phase starts when the relay is closed.

Generator start/stop will still show this as "Warm-up", though. So the phase may be a bit longer than the user configured, because it took the genset x seconds to be started.

mpvader, 2024-10-15:
To make sure I understand it:

am I correct that this issue only affects connected gensets, and relay gensets with digital input?

and that the solution affects all three types?

(I'm fine with the solution)

ReinvdZee, 2024-10-15:
As discussed: We will remove the dependency of the warm-up timer on the feedback mechanism (genset status code or digital input). An incorrectly wired digital input would prevent the multi from drawing power from the generator as it will hang in the warm-up phase (because that will only end when the generator is started).

Instead, we'll make an independent timer which will just time the warm-up phase regardless of whether the generator is actually running. After the timeout, it'll proceed to the 'running' state and the multi will pull power from the generator. In this case, an incorrectly wired digital input will only lead to runtime not being counted. In addition, this also makes sure that the system will always be in 'warm-up' for the time configured.

closed 2024-10-18 by jhofstee