victronenergy/venus #1377: MQTT connection status paths Mqtt/Bridges/* incorrect

wiebeytec, 2024-10-28:
I just saw an installation that said this on dbus:

```
 'Mqtt/Bridges/GXdbus/Connected': [],
 'Mqtt/Bridges/GXdbus/ConnectionStatus': 'deactivated',
 'Mqtt/Bridges/GXrpc/Connected': [],
 'Mqtt/Bridges/GXrpc/ConnectionStatus': 'deactivated',
```

While FlashMQ thought this:

``` 
$SYS/broker/bridge/GXdbus/connection_status Connected
$SYS/broker/bridge/GXdbus/connected 1
$SYS/broker/bridge/GXrpc/connection_status Connected
$SYS/broker/bridge/GXrpc/connected 1
```

The last note about the bridges in the FlashMQ log was about a reload with seemingly an unchanged config. So, that's the first thing to try.

Venus OS version v3.50~39.

mpvader, 2024-10-29:
@wiebeytec this is still in the category “not confusing to users” so no need to hold the release over this?

wiebeytec, 2024-10-29:
I haven't figured out yet if this can also mean it shows confusing info about actual connection problems.

closed 2024-10-29 by jhofstee