victronenergy/venus #1379: dbus-modbus-client: signal 11

dirkjanfaber, 2024-10-29:
Seen on site https://vrm.victronenergy.com/installation/274468/overview. Several signal 11 errors in the output of `dmesg`:

```
[395466.683303] CPU: 0 PID: 17862 Comm: dbus-modbus-cli Tainted: G           O      5.10.109-venus-17 #1
[395466.692760] Hardware name: Allwinner sun7i (A20) Family
[395466.698157] PC is at 0x4a4f17e8
[395466.701509] LR is at 0x1c0c598
[395466.704713] pc : [<4a4f17e8>]    lr : [<01c0c598>]    psr: 000f0010
[395466.711112] sp : 01c0c598  ip : 00000000  fp : 00000000
[395466.716428] r10: 00000000  r9 : 01d342e4  r8 : 01d27ec0
[395466.721878] r7 : 000000af  r6 : 01d2c770  r5 : beafe478  r4 : 0000000b
[395466.728500] r3 : 00000008  r2 : 00000000  r1 : 01c0c598  r0 : 00000000
[395466.735173] Flags: nzcv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
[395466.742595] Control: 10c5387d  Table: 4364406a  DAC: 00000055
[395466.748500] CPU: 0 PID: 17862 Comm: dbus-modbus-cli Tainted: G           O      5.10.109-venus-17 #1
[395466.757715] Hardware name: Allwinner sun7i (A20) Family
[395466.763070] [<c010d19c>] (unwind_backtrace) from [<c010a2c8>] (show_stack+0x10/0x14)
[395466.770918] [<c010a2c8>] (show_stack) from [<c0658fa8>] (dump_stack+0x98/0xac)
[395466.778239] [<c0658fa8>] (dump_stack) from [<c01342a0>] (get_signal+0x794/0x79c)
[395466.785724] [<c01342a0>] (get_signal) from [<c0109984>] (do_work_pending+0x100/0x550)
[395466.793640] [<c0109984>] (do_work_pending) from [<c01000cc>] (slow_work_pending+0xc/0x20)
[395466.801894] Exception stack(0xc5361fb0 to 0xc5361ff8)
[395466.807032] 1fa0:                                     00000000 01c0c598 00000000 00000008
[395466.815292] 1fc0: 0000000b beafe478 01d2c770 000000af 01d27ec0 01d342e4 00000000 00000000
[395466.823552] 1fe0: 00000000 01c0c598 01c0c598 4a4f17e8 000f0010 00000002
```

The logfile in `/data/log/dbus-modbus-client/` shows 2 segmentation faults and a lot of timed out errors. 

```
2024-10-29 01:20:48.272715500 INFO     registered ourselves on D-Bus as com.victronenergy.evcharger.evc_HQ2241QZR67
2024-10-29 01:20:48.607393500 INFO     registered ourselves on D-Bus as com.victronenergy.grid.ve_HQ2414ZZPCN
2024-10-29 01:21:54.050530500 INFO     [udp:192.168.178.87:502:1] Device failed: Error reading registers 0x3080-0x308b: Modbus Error: [Input/Output] timed out
2024-10-29 01:21:54.227334500 INFO     [udp:192.168.178.87:502:1] Found Energy meter: Victron Energy VM-3P75CT
2024-10-29 01:21:54.365423500 INFO     registered ourselves on D-Bus as com.victronenergy.grid.ve_HQ2414ZZPCN
2024-10-29 01:22:39.395705500 INFO     [udp:192.168.178.87:502:1] Device failed: Error reading registers 0x3032-0x3056: Modbus Error: [Input/Output] timed out
2024-10-29 01:22:39.574238500 INFO     [udp:192.168.178.87:502:1] Found Energy meter: Victron Energy VM-3P75CT
2024-10-29 01:22:40.068027500 ERROR    [udp:192.168.178.87:502:1] Error reading register 0x1009: Modbus Error: [Input/Output] timed out
2024-10-29 01:22:43.086312500 INFO     [udp:192.168.178.87:502:1] Found Energy meter: Victron Energy VM-3P75CT
2024-10-29 01:22:43.298004500 INFO     registered ourselves on D-Bus as com.victronenergy.grid.ve_HQ2414ZZPCN
2024-10-29 01:24:35.204445500 Fatal Python error: Segmentation fault
2024-10-29 01:24:35.204454500 
2024-10-29 01:24:35.204455500 Thread 0xb51e8460 (most recent call first):
2024-10-29 01:24:35.204457500   File "/usr/lib/python3.8/site-packages/dnslib/bit.py", line 50 in get_bits
2024-10-29 01:24:35.204461500   File "/usr/lib/python3.8/site-packages/dnslib/label.py", line 236 in decode_name
2024-10-29 01:24:35.204464500   File "/usr/lib/python3.8/site-packages/dnslib/dns.py", line 802 in parse
2024-10-29 01:24:35.204467500   File "/usr/lib/python3.8/site-packages/dnslib/dns.py", line 109 in parse
2024-10-29 01:24:35.204653500   File "/opt/victronenergy/dbus-modbus-client/mdns.py", line 115 in run
2024-10-29 01:24:35.204657500   File "/usr/lib/python3.8/threading.py", line 870 in run
2024-10-29 01:24:35.204660500   File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
2024-10-29 01:24:35.204663500   File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap
2024-10-29 01:24:35.204666500 
2024-10-29 01:24:35.204667500 Thread 0xb5c87460 (most recent call first):
2024-10-29 01:24:35.204694500   File "/opt/victronenergy/dbus-modbus-client/watchdog.py", line 24 in run
2024-10-29 01:24:35.204697500   File "/usr/lib/python3.8/threading.py", line 870 in run
2024-10-29 01:24:35.204700500   File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
2024-10-29 01:24:35.204703500   File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap
2024-10-29 01:24:35.204706500 
2024-10-29 01:24:35.204707500 Current thread 0xb6f43010 (most recent call first):
2024-10-29 01:24:35.360271500   File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 497 in run
2024-10-29 01:24:35.360279500   File "/opt/victronenergy/dbus-modbus-client/dbus-modbus-client.py", line 435 in main
2024-10-29 01:24:35.360283500   File "/opt/victronenergy/dbus-modbus-client/dbus-modbus-client.py", line 438 in <module>
2024-10-29 01:24:35.531599500 *** starting dbus-modbus-client ***
```

System is running v3.50~37, Energy meter is running lastest firmware version v1.08.

mansr, 2024-10-29:
This is a know problem that occurs sporadically with no discernible trigger. Hopefully it will go away with a newer version of something. Meanwhile, the impact is minimal as the process is automatically restarted.

The timeouts suggest that the network is unreliable. Suggest switching from wifi to wired Ethernet.

mpvader, 2024-10-29:
ok thanks for the quick reply - closing