victronenergy/venus #1383: Zero feed with fronius on pv out

pahorpet, 2024-11-04:
If there is zero feed in Active and "Individual phase" in ESS, the fronius is still throttling to sum not to the lowest phase that mean there is overshoot in some phase. That mean delivery to the grid. What is forbiden in that setting.

mpvader, 2024-11-06:
hi @pahorpet , thank you for the report, we'll look into this

closed 2024-11-07 by izak

pahorpet, 2024-11-07:
As you wrote it is old issue why not to fix yet? Could you increase the priority for that. Thank you.

izak, 2024-11-07:
Hi @pahorpet , it's not a very easy issue to fix. What we have to do is limit the PV-inverter to three times the phase with the lowest consumption, which will actually cause the other two phases to start consuming power, and in an ESS system that will discharge the battery. This will cause charge/discharge cycles on the battery, or it might cause PV to be lost (which could have been used to charge the batteries), which unfortunately makes the whole thing quite tricky. Combined with several other things taking up our time, and this feature actually not being asked for very often, it got back-burnered. But we will get to it. Messages like yours are important: That is how we know more people are looking for the feature, and if there are two, then there are probably 20.

pahorpet, 2024-11-10:
Hi @izak, thank you for your response.
As you wrote that would charge / discharge the battery I think that this is the case what I was looking for as when you activate Individual phases you will already get a note that is is not the most efficience way to hande the power, but there is no mention that PV production on AC is not reguleted. I have fixed that for now on some instalation via nodered but that is temporary fix, hope this issue will be solved soon. If I could help somehow (testing, optimazing the feedback loop,...), just let me know.

Thank you 
Your installer PVolt

jandvorak471, 2025-02-26:
It is really needed to limit all Fronius inverter phases to value of phase with the smallest load in that time. Otherwise there is too much energy going to the grid with risk of sanctions in the Czech Republic. So generally in ESS AC coupling with no feed in we need to limit Fronius on base of lowest consumption phase no on base of total energy balance of all phases. Every installation in the Czech Republic with Fronius and no feed in ciondition have/will have problem with current situation of Fronius regulation.