victronenergy/venus #1384: MFD App: Remote Console with classic UI doesn't open?

mpvader, 2024-11-06:
https://community.victronenergy.com/t/v3-50-confusion-with-the-marine-mfd-app/10355

[image]

mman, 2024-11-06:
Interesting, I have quickly cross-checked that it still works as expected, and it indeed does not work on Furuno while working on other MFDs... but it was not like this before, so this must have been broken during one of the recent builds, and if I `curl http://A.B.C.D/gui-v1/` I can indeed see some new JS logic that was not there before...  and that is probably confusing the old WebKit browser in Furuno...

mman, 2024-11-06:
Furuno left, Simard right...

[image]

mpvader, 2024-11-06:
nice find, pls see if you can fix that.

Either let us know what needs fixing, or do it directly by making a PR to where we have that in git, which is most likely here:

https://github.com/victronenergy/javascript-vnc-client/blob/master/index.php

or here:

https://github.com/victronenergy/venus-www

mman, 2024-11-06:
Although I could not yet confirm precisely the root cause, as it is a bit tricky to install `window.onerror` handlers inside `iframe` that HTML5 app uses to load remote console, from my existing experience I speculate that with very high probability this is the root cause:

The `async function` syntax very probably causes syntax error on Furuno, thus stopping the `window.onscriptload` handler from executing, thus getting frozen.

https://github.com/victronenergy/javascript-vnc-client/blob/cd421dc32bcf06c357d04d30f70ba241e716279d/index.php#L209-L228

Introduced by this commit a month ago: https://github.com/victronenergy/javascript-vnc-client/commit/4923727ba2c745e97d2fef1285caea7ac6549a48

Solution: those two methods need to be rewritten to plain old promise based JS syntax.

mman, 2024-11-06:
Just to confirm, if I `mount -o remount,rw /` and modify `/var/www/venus/gui-v1/index.php` to revert the commit introducing the `async function` it works again.

mman, 2024-11-06:
Here is the fix: the method `async function isGuiv2Running` can be dropped entirely, and the other method can be rewritten like this:

Although I am not entirely sure what the purpose of that method is so I could not test whether it works as expected. But it definitely can be compiled by WebKit on Furuno and allows the classic console to load again.

```
function checkGuiv2() {
      fetch(location.protocol + "//" + location.hostname, {cache: "no-cache"})
        .then(function(response) {
            if (response.ok) {
                if (response.redirected && response.url.includes("gui-v2")) {
                    location.replace(response.url)
                }
            }
        })
        .catch(function(error) {})
}
```

jhofstee, 2024-11-07:
The fix is included in master and backported to v3.52~2.

closed 2024-11-07 by jhofstee