victronenergy/venus #1386: gui-v1: add new temperature types

mpvader, 2024-11-09:
* Gui-v1: https://github.com/victronenergy/gui/blob/4a3177ca4be9851afb3e0af214f9cdd29aca9e9b/qml/PageTemperatureSensorSetup.qml#L20
* Gui-v2: https://github.com/victronenergy/gui-v2/blob/0d642160755e7b80bc862a4f20af6b40d2e13bc4/src/enums.h#L491

I'm not 100% sure why these were added to gui-v2, perhaps/probably to have different temperature ranges per type. Anyway, to move forward lets add them to gui-v1 as well.

Note [that there is this issue as well that for Ruuvis and such](https://github.com/victronenergy/venus/issues/1385) its now not possible to configure any of the new types, so use local temp inputs on a GX for testing this fix.

dirkjanfaber, 2024-11-09:
Added to [dbus_modbustcp](https://github.com/victronenergy/dbus_modbustcp/tree/faberd/temperature-types)
and to [node-red-contrib-victron](https://github.com/victronenergy/node-red-contrib-victron/commit/874d76400437c29761ff866f930d1172bfd40ee9)

But not yet in a release of the packages.

mpvader, 2024-11-12:
Can be closed @jhofstee ?

jhofstee, 2024-11-12:
for gui-v1, yes. But the changes of Dirk-Jan are not included yet. So I left it open for now, it either needs to be renamed and milestones updated or we need to create a new issue for that, so we won't forget it.

dirkjanfaber, 2024-11-12:
Both the dbus_modbustcp and node-red-contrib-victron are in their master branches. So for me this issue can safely be closed.

jhofstee, 2024-11-12:
venus issue are normally closed after the fix are included in venus and ready for testing. The reason to do that is that while you might think it is fixed in your master branch / latest release, but if that release never gets shipped, users still have the issue.

dirkjanfaber, 2025-04-16:
Included in [v1.0.76](https://github.com/victronenergy/dbus_modbustcp/commit/cd9284e956cd26343ecfb00670d664b7af9a5287) of dbus_modbustcp and [v1.6.1](https://github.com/victronenergy/node-red-contrib-victron/releases/tag/v1.6.1) of node-red-contrib-victron, which made the todo a long time ago and got shipped. Closing this (and will be adding a remark to close the attached issue for future issues)

closed 2025-04-16 by dirkjanfaber