victronenergy/venus #1390: Remote veconfigure: Occasional configuration fails with error 46 (uploaded file does not match model)

vejpasop, 2024-11-15:
Bug since Venus V3.50 (reported on V352~4)

[image]

Log shows 

```
2024-11-14 07:59:27.644259500 INFO:vrmmk2vsc:write_settings from file /data/vrmfilescache/331d4925c9988313892aa75223ae1b88_331d4925c9988313892aa75223ae1b88.rvsc
2024-11-14 07:59:27.679613500 INFO:__main__:Starting vebusremoteconfig response
2024-11-14 08:00:11.607149500 043.947 ERR.vscFile: Settings info section does not contain data belonging to device type 2
```
The file however does contain data for device type 2. 

A re-attempted remote configuration completed successfully.

vejpasop, 2025-02-27:
Fixed in mk2vsc V1.31

fionnb, 2025-03-05:
Aaaand ... where do I get that binary? :-) 
Latest official firmware only includes 1.30
Currently unable to configure my newly installed MKII due to this bug!
No, I have no cable.

mpvader, 2025-04-16:
Already in the Venus OS v3.60 beta series.

And will be part for Venus OS v3.56 official, expected in coming two weeks.

closing.

closed 2025-04-16 by mpvader