victronenergy/venus #1396: Default visibility setting of battery monitors in "Battery measurements"

ReinvdZee, 2024-11-19:
When clicking the battery drilldown, the user is redirected to a list of battery monitors:
[image]

Available battery monitors can be set to 'visible' or 'hidden' in system setup -> Battery monitors:

[image]

By default, all battery monitors are hidden, except the active battery monitor. We should improve this by enabling 'dedicated' battery monitors (smartshunts, BMSes, etc.) by default. 

The setting to show or hide a battery monitor is e.g. `Settings/SystemSetup/Batteries/Configuration/com_victronenergy_battery/2/1/Enabled`

ReinvdZee, 2025-01-03:
Systemcalc defines several types of battery trackers, each corresponding to a certain service providing battery data. I'd say the `com.victronenergy.battery` service is the only one dedicated battery monitor. For this type, both a `BatteryTracker` and a `SecondaryBatteryTracker` [are created](https://github.com/victronenergy/dbus-systemcalc-py/blob/master/delegates/batterydata.py#L219-L220). The first is the main battery tracker, and the second tracks `/Dc/1/Voltage`, e.g. the starter battery voltage. 

For example, the smartshunt has both trackers:
[image]

We can enable the visibility of the main tracker such that only the main tracker of `com.victronenergy.battery` services is visible by default (like in the screenshot above). I [drafted](https://github.com/victronenergy/dbus-systemcalc-py/commit/2667c591fe140fe225fb88dc6406f904d1d65d53) an implemenation for this.

@izak @mpvader Is this the desired behavior?

mpvader, 2025-04-16:
already in there

closed 2025-04-16 by mpvader