victronenergy/venus #1399: Ruuvitag Data Spikes

askpatrickw, 2024-12-05:
I see these spikes in my Ruuvitag data.

[image]

I found a 2022 related thread, but nothing in this repo about a fix.

https://communityarchive.victronenergy.com/questions/119966/spikes-in-data-collected-from-ruuvitags.html

askpatrickw, 2024-12-05:
I updated the description to note that I'm pulling this data into Home Assistant over MQTT.

askpatrickw, 2024-12-07:
A few of us over on the Ford Transit USA Forum have noticed it is not just Ruuvi, but any values collected over MQTT.

I see this in the SOC for example.

[image]

askpatrickw, 2025-04-16:
I don't think which sensor (in HA terminology) matters.

askpatrickw, 2025-04-16:
And I'm evening rounding the value in the case of SOC.

```json
  - name: "Battery SOC"
    unique_id: "HX4MNJHAXmW_bkFhrQPYt"
    device_class: battery
    state_topic: "victron/N/c0619ab1f84c/system/0/Dc/Battery/Soc"
    suggested_display_precision: 0
    value_template: "{{ value_json.value|round(0) }}"
    unit_of_measurement: "%"
    device:
      {
        identifiers: ["Cerbo GX"],
        manufacturer: "Victron Energy",
        model: "Cerbo GX",
        name: "Cerbo GX",
      }
```

dirkjanfaber, 2025-04-16:
@mpvader : We could track the SOC and temperature value of that system via docker/grafana. If that shows spikes, we know the cause is somewhere in mqtt. Otherwise the problem must be on the HA side.

dirkjanfaber, 2025-04-16:
@askpatrickw : I've set up the mqtt logging on our side. Please report future spikes with the (approximate) timestamp in this call. It might still take a  while before we can look into it, but at least we know where to look then.

askpatrickw, 2025-04-16:
@dirkjanfaber Not sure I follow. How will this logging work exactly?

Maybe you mean you turned that on in VRM?
I don't use VRM...

dirkjanfaber, 2025-04-16:
Indeed. The site is setup to log to VRM, even if you don't use that. Of course you can switch that off, but as it might help us find the cause for the spikes, I'd like you to leave that on while this issue is still open.

askpatrickw, 2025-04-17:
@dirkjanfaber OK... actually it has been on.
Around 6pm Pacific there are bottom spikes.

dirkjanfaber, 2025-04-29:
Is it possible for you to set the _VRM Online portal -> VRM Portal_ value to ***Full*** for a while? 

[image]

askpatrickw, 2025-04-30:
@dirkjanfaber I've turned it on.

dirkjanfaber, 2025-05-09:
@askpatrickw  I've been logging for the last 2 days, but don't notice any spikes in the mqtt data. Have you seen any spikes lately?