victronenergy/venus #1400: (re)connect CCGX to hidden WiFi SSID

Subnum12, 2024-12-07:
As requested from others since quite a long time I it would be nice to follow RFC and allow connection to hidden WiFi SSID.

https://community.victronenergy.com/questions/35671/can-the-ccgx-connect-to-a-hidden-wifi-network.html

Anyway, even that can concluded somehow as feature request, not reconnecting to network previously configured would be a bug.

The workaround (make SSID visible, configure GX-device with WiFi, make SSID hidden again) does work sometimes. However, often GX-device does not reconnect to the configured WiFi once it is hidden. If it would at least try to connect to configured networks (even if they are not visible) that could work for most users I guess.

jhofstee, 2025-02-07:
On a ccgx, you should be able to connect by its MAC-address.

itsmorefun, 2025-02-07:
Why close when the problem still exists?